solokeys/solo1 #353: use Nucleo32 board,  Do I have a problem doing this？？

meeos, 2020-01-05:
I read https://docs.solokeys.io/solo/nucleo32-board/

I buy Nucleo32 but I test failed (on https://webauthn.org/)

[image]

first step:

Execute the following command and the result is shown in Fig

`cd targets/stm32l432 && make make cbor && make build-hacker`

[image]

two step：

all.hex we got before and use all.hex to flashing with STM32CubeProgrammer tools 

[image]

meeos, 2020-01-05:
I test failed 
[image]

jolo1581, 2020-04-04:
I also use the Nucleoboard.
[image]

Works fine under Linux.
Do you follow the instructions on Solo documentation
[https://docs.solokeys.io/solo/nucleo32-board/](https://docs.solokeys.io/solo/nucleo32-board/) 

Remember: the USB connection of Solo is not the Connection of ST-Link

alinelena, 2020-05-22:
I have more or less the same problem as @meeos . 
I read in programming page https://github.com/solokeys/solo/blob/master/docs/solo/building.md
```
There exists a development board NUCLEO-L432KC you can use; The board does contain a debugger, so all you need is a USB cable (and some udev rules).
```
NUCLEO-L432KC is the device in your pictures too...
however if one goes to the specific page for the board

https://github.com/solokeys/solo/blob/master/docs/solo/nucleo32-board.md

```
Board does not provide an USB cable / socket for the target MCU communication. Own provided USB plug has to be connected in the following way:
```
in the table header the one gets pin/arduino pin d10/pa11 thrown in... without any explanation how arduino gets into the play.

Then later the button needs to be connected to PA0... there is no PA0 button on the board... this assuming that we speack about nucle-L432KC.

so any help is appreciated.

nickray, 2020-05-22:
We can't really support self-directed studies here. But maybe https://os.mbed.com/media/uploads/bcostm/nucleo_l432kc_2017_10_09.png from https://os.mbed.com/platforms/ST-Nucleo-L432KC/ is helpful to get started with pins (should also be in package you bought).

alinelena, 2020-05-22:
@nickray thanks I understand what you say 

long story short make flash works fine when connected to "normal" usb. 
somehow I can access the key only via "debug" and with Vin connected to the 5V.
like this I could test it and managed to verify it.

however seems not to boot the solo firmware when connected directly via "normal" usb... 
I wonder if there are only messed up udev rules.

jolo1581, 2020-05-23:
@alinelena 
In Linux you might Setup udev rules for ST-Link.
See #418

jolo1581, 2020-05-23:
@alinelena:

Here are some pictures:
**USB Connection**
[image]

**USB Powered without ST-Link cable**
In case using Solo bootloader, you can use the 5V from USB port to power the Nucleo board. But make sure, if you want to program your device with ST-Link instead of Solo bootloader you **must not connect** USB 5V to VIN.
[image]

**LED connection**
[image]

[image]

**Button**
[image]

If you already programmed all.hex (Application + Bootloader) over ST-Link, you can use Solo bootloader for programming instead of ST-Link.

alinelena, 2020-05-23:
@jolo1581 thank you very much your new pictures are much better for beginners. The leds are cool but I think I will give them a miss for now.

I have reversed your previous picture and things worked. I tested both methods of burning the firmware and worked...

once things got clear in my mind process was quite easy.

at least I have managed to pass the tests in browser and 
solo key commands work too.
Yes I have setup the udev rules for the stm and solo. 
I have reached up to #441

meeos, 2020-05-27:
@nickray  @alinelena    I read the source code of solo, 
Power supply from others micro-usb of st-link   then realization of virtual serial communication  from PA11and PA12 pin ，use double usb port

jolo1581, 2020-05-27:
@meeos 
Can you show us your setup?
The VCP is dedicated on PA2 and PA15. Refer schematic of nucleo board.

The Solo USB port is D+ on PA12 and D- on PA11. But this is no VCP

meeos, 2020-06-11:
@jolo1581 I have solved this problem， need double usb ,one for power supply , other usb link pa11 and pa12  thinks very much

closed 2020-06-11 by meeos

vaporwave9, 2021-05-02:
@jolo1581
Inbound thread necro, sorry in advance:
Q1:
Are the pinouts in your illustrations here https://github.com/solokeys/solo/issues/353#issuecomment-633023517 identical to the real product I can purchase?

Q2:
And if so, what firmware would I flash to have the real deal? Is flashing firmware-4.1.2.hex into a NUCLEO devboard from Reichelt with STM32CubeProgrammer enough for me?

The reason why I am not purchasing a commercial solution is that I want to be able to do backups myself, i.e.:
https://dmitryfrank.com/articles/backup_u2f_token#caveat_with_the_counter

Q3:
Since the hacker version is not for daily use, I assume my alternative would provide me an option to both have access to my data over USB2 and the built-in ST-Link and still use it over USB1 safely without being hacked?

jolo1581, 2021-05-02:
@vaporwave9 

> Inbound thread necro, sorry in advance:
> Q1:
> Are the pinouts in your illustrations here https://github.com/solokeys/solo/issues/353#issuecomment-633023517 identical to the real product I can purchase?

Yes

> 
> Q2:
> And if so, what firmware would I flash to have the real deal? Is flashing firmware-4.1.2.hex into a NUCLEO devboard from Reichelt with STM32CubeProgrammer enough for me?

Think you also need a bootloader.
Maybe @conorpp can give the answrt

> 
> Q3:
> Since the hacker version is not for daily use, I assume my alternative (regular firmware) would provide me an option to both have access to my data over USB2 (connected to the built-in ST-Link) and still use it over USB1 (connected to work-PC) safely without being hacked?

I don't understand your idea. Maybe you can give more information

vaporwave9, 2021-05-02:
@jolo1581 
Thanks for the fast answer!

> I don't understand your idea. Maybe you can give more information
> 

The way I understand it, the NUCLEO has two USB interfaces for the actual chip and another for the ST-Link programmer on the backside. If I purchase a Solokey it can be updated over the regular port since it already has its bootloader and doesn't need an extra ST-Link anymore like the development board. Through webservices even: https://update.solokeys.com/
I suspect for safety the regualr bootloader must be "bolted down" somehow so other people cannot read or re-flash it.  The hacker version seems to have an open bootloader that would allow access to data over USB1?

My idea was flashing the bolted down production ready firmware with the secure bootloader, but still be able to dump the contents as backup.hex over the ST-Link on the development board? That way USB1 connected to the ARM cortex is "safe" but I can do backup.hex over USB2 of the ST-Link?

You are the expert :smile:  , basically I just assumed naively that I can use two USB at the same time, with different levels of security clearance to the devboard.

> Yes
> 

Many thanks, then I can flash without worrying to change pin assignments :)

> Think you also need a bootloader.
> Maybe @conorpp can give the answrt
> 

Oh I was thinking the firmware.hex was like the "all.hex" that was mentioned in this thread. Is there no single file I can flash? I'm at a loss which of the many bootloaders offered to use? Whilst I have programmed numerous microcontrollers, i never had to worry about bootloaders. Espressif is sort of "easy mode" for casual-tier MCU enjoyers like me.

vulpes9, 2021-05-04:
Alright, many unknowns and open questions answered itself by try-and-error. Sadly I could not use my RGB LEDs as I realized too late that there is a common VDD in the sketch and my LEDs only have a common VSS.

It's a bit of a dusty mess under my screen and a prototype board twice the sice will sit there now, but I love it!
All works and creating backups as well, many many thanks again!

[image]