solokeys/solo1 #354: Provide two devices like Yubico

ceving, 2020-01-06:
This is an enhancement request.

It would be useful for SoloKeys to provide two devices like Yubico.

Yubico provides two devices:

    hid-generic 0003:1050:0116.0002: input,hidraw1: USB HID v1.10 Keyboard [Yubico Yubikey NEO OTP+U2F+CCID] on usb-0000:00:15.0-8/input0
    hid-generic 0003:1050:0116.0003: hiddev0,hidraw2: USB HID v1.10 Device [Yubico Yubikey NEO OTP+U2F+CCID] on usb-0000:00:15.0-8/input1

But SoloKeys provides just one device:

    hid-generic 0003:0483:A2CA.0001: hiddev0,hidraw0: USB HID v1.11 Device [SoloKeys Solo 3.0.0] on usb-0000:00:15.0-3/input0

This is problematic in virtual environments. In order to pass through a USB device to a guest OS, it is necessary to have a device. If the host takes the device as an input device, there is nothing left, which can be passed to the guest. Yubico is less problematic, because the host can take the input device and the other device can be passed to the guest.

My1, 2020-01-14:
well Yubico doesnt provide 2 devices for fun. the entire OTP interface runs on the Keyboard. the Solo doesnt have that?

also how do you pass other things that dont spawn multiple USB devices?

tERyceNzAchE, 2020-01-15:
Would this allow (eventual) support for a static password on button press (like YubiKey) ?
I know there is no plan to implement this feature, but would providing a keyboard device make this possible?

conorpp, 2020-01-15:
This probably won't happen on the current hardware, but very likely on the new Solo revision being worked on.

nickray, 2020-01-15:
This is not a good issue, as the description is incorrect/misleading ;)

tERyceNzAchE, 2020-01-15:
This is a GREAT issue, with a misleading title/description.  :)

My1, 2020-01-15:
I think differently, also I think this is an issue with your hypervisor as I have passed Solos using Virtualhere (granted not a hypervisor, but besides the point) to other computers more than often enough flawlessly