solokeys/solo1 #357: what is the ctap_encode_der_sig() used for?

bling206, 2020-01-13:
Hi 

I am tracing about the solo's source code.
Why does the signature need to be used after the ctap_encode_der_sig() from instead of using directly the output from crypto_ecc256_sign().

Thank you!

nickray, 2020-01-13:
There are two ways to serialize NIST P-256 signatures: encoding (r, s) as consecutive 32-byte arrays interpreted as big-endian integers (this is what `micro-ecc` outputs), or ASN.1 DER format ("SEQUENCE" of two, followed by two "INTEGER"s), which is what this encoding function does. CTAP uses COSE (https://tools.ietf.org/html/rfc8152), which specifies the latter format.

nickray, 2020-01-13:
"Well actually" :D 
https://github.com/w3c/webauthn/issues/799 I guess at some point the intent may have been to use COSE, which would specify the raw format, whereas by "ossification" ASN.1 DER needs to be used.

bling206, 2020-01-14:
Got it!! Thank you very much!!

jolo1581, 2020-03-29:
Can this issue be closed?

jolo1581, 2020-03-29:
Think we can close this issue, right?

closed 2020-03-29 by nickray