solokeys/solo1 #360: Somu hacker to locked transition problem

NicoPy, 2020-01-16:
Hi,
I received my Somu 3 keys pack :)
Looks like I made an error when I ordered them since I received the hacker version.
So, I decided to convert them to locked mode myself.
I tried on one key first. 
Not so easy.
I followed the instructions described [here](https://docs.solokeys.io/solo/programming/#updating-a-hacker-to-a-secure-solo).
My PC runs Windows 10. I had hard time installing libusb. Windows prevented the driver installation. I found the solution [here](https://www.smallcab.net/download/programme/xm-07/how-to-install-libusb-driver.pdf). In this document, at page 14, it is specified to not click "Install Now..." button . I clicked it and all gone well. The driver is now installed.

So, Here what I did :
- _solo program aux enter-bootloader_
- _solo program aux enter-dfu_
- unplug then replug the key
- _solo program dfu bundle-secure-non-solokeys-3.0.1.hex_
- _solo program aux leave-dfu_
After following the above procedure, it look like my key can't go to DFU mode anymore which is intended. 
BUT... 
_solo key version_ says :
3.0.1 unlocked

_solo key verify_ says :
Please press the button on your Solo key
Register valid
Solo Hacker firmware.

_solo key update --secure_ gives an error

I tried _solo key update_ one more time but this did not help.

Is this a bug in solo utility reporting bad version ?
Does this mean the convertion procedure is not corect ?
What is the status of this key ? Locked, unlocked, mixed ?

thomiel, 2020-01-20:
I think, I have the same problem on Linux, locking my Somu to secure. I followed the guide at 'Programming Instructions', after I discovered that the solo key update '--secure' option obviously was removed, but get this exception:

```
solo program dfu ~/Downloads/bundle-secure-non-solokeys-3.0.1.hex 
Traceback (most recent call last):
  File "./solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/solo/cli/program.py", line 63, in dfu
    dfu = solo.dfu.find(serial, attempts=connect_attempts)
  File "/home/thomas/Code/solo/venv/lib/python3.6/site-packages/solo/dfu.py", line 36, in find
    raise Exception("no DFU found")
Exception: no DFU found
```

It seems like there is either a bug in the software or the documentation is incomplete. Any ideas?

kuba86, 2020-02-05:
Hi, I had the same problem, in documentation I did not notice the first time info: 
`Windows users need to install libusb for solo-python to work with Solo's DFU.`

how to:

1. Install solo-python in Python venv
2. Run CMD with admin rights and activate the venv (scripts\activate.bat)
3. In CMD `solo ls && solo key version && solo key verify && solo key update` (should update to newest version) 
4. once updated run in CMD `solo program aux enter-bootloader && solo program aux enter-dfu`
5. unplug and plug once again Somu (it should NOT be blinking or anything)
6. download libusb-win32-bin-1.2.6.0.zip file from https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/1.2.6.0/ extract it
7. run **with admin rights** libusb-win32-bin-1.2.6.0\bin\inf-wizard.exe
8. find STM32 BOOTLOADER. and click NEXT, NEXT, INSTALL NOW. 
9. Once installed in CMD `solo program dfu bundle-secure-non-solokeys-X.X.X.hex` 
10. you should see something like that:
`erasing...
downloading 100.00%  08038800 - 0803882e ...
time: 5846 ms
verifying...
reading 100.00%  08038800 - 0803882e ...
firmware readback verified.`
11. in CMD `solo program aux leave-dfu`
12. unplug and plug once again Somu - it should have green LED

now.... When I do that and run in CMD `solo ls && solo key version && solo key verify`

I still see:

`:: Solos
XXXXXXXX: Solo 3.0.1
3.0.1 unlocked
Please press the button on your Solo key
Register valid
Solo Hacker firmware.`

However, when I try to repeat the process, Somu will not enter `dfu` My assumption is that it means it is now converted to Secure. 

can anybody confirm if after locking or converting Somu from Hacker to Secure `solo key version` and `solo key verify` would show it as Hacker or Secure?

conorpp, 2020-02-12:
I recently discovered that the `bundle-secure-non-solokeys-3.0.1.hex` was not built with the proper lock flag, and did set the lock config on the device.  This has been fixed with the latest 3.1.0 bundle build.  But since it does use the verifying bootloader, it does prevent unauthorized updates and any DFU updates.

Apologies for the mistake, and resulting confusing status reported by `solo-python`, should be resolved now.

NicoPy, 2020-02-13:
@conorpp So, with hacker mode keys, converting them to locked mode with 3.1.0 bundle should be OK.
But, what should be done with the "badly" converted keys ?

conorpp, 2020-02-13:
I just released a patch (`3.1.1`) that adds a command that will force lock the flash.  You can fix the "badly" converted keys by doing the following.

```
# update to 3.1.1
solo key update

# Trigger the new flash lock command
python3 -c "import solo ; d = solo.client.find() ; d.send_data_hid(0x99 ^ 0x80,'')"
```

conorpp, 2020-02-13:
Also, need to reboot the key afterwards :)

closed 2020-02-22 by conorpp

NicoPy, 2020-02-23:
`python -c "import solo ; d = solo.client.find() ; d.send_data_hid(0x99 ^ 0x80,'')"`
-In normal mode, the command does nothing.
-In update mode, the command does not work (using Windows 10 with Python 3.8.0-64bit). I get the following error :
```
Not using FIDO2 interface.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Python38\lib\site-packages\solo\client.py", line 142, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "C:\Python38\lib\site-packages\fido2\hid.py", line 94, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND
```
The firmware is correctly updated :
```
solo key version
3.1.1 locked
```
Why `d.send_data_hid(0x99 ^ 0x80,'')` ? Why not `d.send_data_hid(0x19,'')` which is the same. Is there a typo ?

NicoPy, 2020-02-23:
Also, with a standard SOLO key updated to 3.1.1 :

In normal mode :
```
solo key version
3.1.1 locked

solo key verify
Please press the button on your Solo key
Register valid
Valid Solo with firmware from SoloKeys.
```

In update mode :
```
solo key version
Not using FIDO2 interface.
Traceback (most recent call last):
  File "c:\python38\lib\site-packages\solo\client.py", line 193, in solo_version
    return self.send_data_hid(0x61, b"")
  File "c:\python38\lib\site-packages\solo\client.py", line 142, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "c:\python38\lib\site-packages\fido2\hid.py", line 94, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\python38\lib\runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python38\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python38\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\python38\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\python38\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python38\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python38\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\python38\lib\site-packages\solo\cli\key.py", line 408, in version
    res = solo.client.find(serial, udp=udp).solo_version()
  File "c:\python38\lib\site-packages\solo\client.py", line 195, in solo_version
    data = self.exchange(SoloExtension.version)
  File "c:\python38\lib\site-packages\solo\client.py", line 165, in exchange_u2f
    raise CtapError(ret)
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND

solo key verify
Please press the button on your Solo key
Not using FIDO2 interface.
Traceback (most recent call last):
  File "c:\python38\lib\runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python38\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python38\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\python38\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\python38\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python38\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python38\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\python38\lib\site-packages\solo\cli\key.py", line 322, in verify
    cert = solo.client.find(serial, udp=udp).make_credential(pin=pin)
  File "c:\python38\lib\site-packages\solo\client.py", line 235, in make_credential
    attest, data = self.client.make_credential(options, pin=pin)
AttributeError: 'NoneType' object has no attribute 'make_credential'
```

```

conorpp, 2020-02-23:
Since the "locked" status is now present, the command worked.  The command I gave does not have any output.  Yes, the command is only available in the 3.1.1 application, and not the bootloader.  Both `d.send_data_hid(0x99 ^ 0x80,'')` and `d.send_data_hid(0x19,'')` would work.

NicoPy, 2020-02-24:
> Since the "locked" status is now present, the command worked. 

Ok. Good to know.

I tried to convert my other SOMU keys to locked mode :
```
solo program aux enter-bootloader
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.

solo program aux enter-dfu
Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)

solo program dfu bundle-secure-non-solokeys-3.1.1.hex
Traceback (most recent call last):
  File "c:\python38\lib\runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python38\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python38\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\python38\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\python38\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python38\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python38\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\python38\lib\site-packages\solo\cli\program.py", line 63, in dfu
    dfu = solo.dfu.find(serial, attempts=connect_attempts)
  File "c:\python38\lib\site-packages\solo\dfu.py", line 36, in find
    raise Exception("no DFU found")
Exception: no DFU found
```
I did not tried to run `solo program aux leave-dfu`since the programming does not work.
My keys do not react to any command.
Are my keys bricked ?

NicoPy, 2020-02-24:
> Since the "locked" status is now present, the command worked. 

Ok. Good to know. 
As the verification answer says `Solo Hacker firmware.` I thought it was wrong.

Thanks a lot for your support Patrick 👍