solokeys/solo1 #362: raspberry pi zero, solo firmware emulation

bmatusiak, 2020-01-18:
this is a stupid idea but, can someone help me 
find a way to emulate the firmware over usb gadget HID on raspberry pi zeroW

here is a repo i have seen, yet to test out, but
https://github.com/Crystalix007/U2FDevice

would be nice attraction for solokey firmware experiments,
the idea is to build the firmware on the piZeroW
and have the device show up as a "ethernet+solo"

this idea is purely for development, 

for a production, there is a lot of stuff to consider and that can be found at https://github.com/Crystalix007/U2FDevice#security-issues

so if someone wants to fix those "security-issues" for production, 
they would need a "HAT" with solokey hardware, and find a way to 
pipe the hardware over something like SPI to the usb_gadget. while also 
allowing updates over the spi. like with https://www.zymbit.com/zymkey/

but i that is a story for a different day.

merlokk, 2020-01-18:
Why gadget? maybe USBIP?
https://github.com/solokeys/openpgp/tree/master/pc

bmatusiak, 2020-01-18:
@merlokk well you can use usb_gadget to emulate devices like @ http://www.isticktoit.net/?p=1383
like ethernet keyboard mass_storage, and i would like to emulate solokey with this

bmatusiak, 2020-01-18:
so i need to start with building a usb report_desc for HID device to match solokey hacker or sorts
and then somehow run solokey firmware instead of usbip use the /dev/hidg0 gadget device

merlokk, 2020-01-18:
you need to create usb descriptors from ones from solokey and use

bmatusiak, 2020-01-19:
found a good article about the report descriptor, 
https://www.rmedgar.com/blog/using-rpi-zero-as-keyboard-report-descriptor

rgerganov, 2020-02-13:
I have already added support for USB HID gadget in Solo, see this [Makefile](https://github.com/solokeys/solo/blob/master/tools/gadgetfs/Makefile). On Ubuntu you can simply run `make install` as root and this will create `/dev/hidg0`. The Solo firmware should be started like this: `./main -b hidg`

On Raspberry Pi with USB OTG you should already have a host controller, so no need to install `dummy_hcd`. Just configure the gadget and you should be good to go.

bmatusiak, 2020-02-13:
will give it a shot and report back with any issues,

renanqts, 2021-04-17:
any chance to make it work in raspberry pi pico?