solokeys/solo1 #363: Somu seems to accept a touch within a few seconds preceeding a request

lov3craft, 2020-01-20:
It looks like a Somu that has been touched within a few seconds prior to a request will automatically validate a request issued after the touch.

Noticed this when playing around with webauthn.io. If I clicked register on the site within approx 2 seconds after touching the key I received a success without having to touch the key AFTER pressing register as I would have expected. Same for login.

Somu seems to cache the touch for a short time despite no pending request.

lf94, 2020-02-16:
Yep, getting the same behavior.

Very specifically:

1. Touch the Somu
2. Click on Register or Login 1 second after

I would suspect this behavior is tied to Somu's button touch polling. We could probably do better.

nekocentral, 2020-03-01:
I also had this happen with my solo secure and hacker on firmware 3.1.1 when logging in on my laptoo