solokeys/solo1 #364: Somu doesn't show any signs of life

notthetup, 2020-01-21:
I received a Somu recently. Yay!

I tried to set it up on my 2019 MBP (USB-C) only with an USB-C -> USB-A adapter. And it wouldn't enumerate at all. Using the `solo` cli I got 

```
>  solo ls
:: Solos
```

I looked at the macOS USB devices list (in `System Information`) and couldn't see it there either.

Next, I tried on an older 2015 Macbook with native USB-A ports, and it had the exact same behavior. 

Wondering if it was a macOS issue, I tried it on a RPi. I got nothing on `lsusb` and nothing interesting happened on dmesg when I plugged it in.

Also, the LED on the device doesn't seem to ever light up.

Any ideas what could be wrong?

patvdleer, 2020-02-14:
I have the same thing, I followed the docs

```sh
git clone --recurse-submodules https://github.com/solokeys/solo
cd solo

VERSION_TO_BUILD=2.5.3
git fetch --tags
git checkout ${VERSION_TO_BUILD}
git submodule update --init --recursive

cd targets/stm32l432
make cbor
make build-hacker
cd ../..

make venv
source venv/bin/activate
solo program aux enter-bootloader
solo program bootloader targets/stm32l432/solo.hex
```

this ran and updated in about 2.5 sec after this I have the same results, nothing in `lsusb` and no leds...

lf94, 2020-02-16:
I am also not getting results on a USB 3 port via `solo ls`.

I've done almost exactly as @patvdleer has done. `lsusb` though **is** reporting STMelectronics device. Trying to debug further on this Saturday night...

lf94, 2020-02-16:
Below is an audit of the investigation I did. I hope this helps someone.

The conclusion for my issue: solved. Root cause: udev rules.
----

Bus 001 Device 015: ID 0483:a2ca STMicroelectronics

solo-python/solo/client.py:62
```
d.descriptor["vendor_id"] == 1155,
d.descriptor["product_id"] == 41674,
```

1155 = 483
41674 = a2ca

Targetted descriptors are fine.

Adding print(solo_devices) to see results.

Try to build solo-python. Builds but decide to look at README.md before running.

README.md:14

[...] ensure you have suitable udev rules in place

I then open docs/solo/udev.md in the solo project.

I try to install the udev rules since Debian Buster does not have the
appropriate rules. I do this by following the instructions in docs/solo/udev.md
which is currently:

```
cd solo/udev
make setup
```

I unplug and replug the Somu, and lo-and-behold:

```
$ solo ls
:: Solos
208F39A4384B: SoloKeys Solo 3.0.0
```

Success.

lf94, 2020-02-16:
Ok, I'm running into new issues that now look closer to @patvdleer 's. No LED and sometimes missing usb entry in `lsusb`.

One way to make it come back is to hold onto the "button" as you plug it in, holding it for around 3 - as you hold, keep running `sudo dmesg`. Do NOT continue if you see DFU mode in the output of `dmesg`. This is a dangerous mode to be in since you can now overwrite more than just the application code and can brick your device.

Once you see your device again on `lsusb`, the LEDs and button itself probably don't work still. This is the point I'm at, but at least you can now continue to flash new application builds onto it.

Currently I'm stuck at that point-

As I'm writing, the Somu is now flashing red and `solo key ls` shows the device...What. I'm getting:

```
SoloKeys Solo Bootloader 3.0.0
```

Which is weird because I built master which is currently 3.1.1?

conorpp, 2020-02-16:
@lf94 it sounds like your key is working.  To produce the same build, you should use the docker build, and on the respective code release.

@notthetup is your key not working?  And the key is in the correct orientation when you plug in (i.e. Somu contacts are meeting USB socket contacts)?  If the light doesn't even turn on from multiple USB-A sockets, you should contact hello@solokeys.com for replacement.

@patvdleer does your LED turn on?  What is your operating system?  Check udev rules for Linux.  Beware of USB-A to USB-C adapters.

notthetup, 2020-02-16:
Thanks @conorpp

I’ve contacted support.

patvdleer, 2020-02-17:
OMFG!

The 3 seconds holding the button did it!

```
[60189.037569] usb 1-2: new full-speed USB device number 60 using xhci_hcd
[60189.186819] usb 1-2: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
[60189.186824] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[60189.186827] usb 1-2: Product: Solo Bootloader 3.0.0
[60189.186829] usb 1-2: Manufacturer: SoloKeys
[60189.186831] usb 1-2: SerialNumber: 209536BB4XXX
[60189.197197] hid-generic 0003:0483:A2CA.0034: hiddev1,hidraw2: USB HID v1.11 Device [SoloKeys Solo Bootloader 3.0.0] on usb-0000:00:14.0-2/input0
```

## some background info:
`sudo dmesg -w` gives the above info.  I removed the udev rules which didn't matter, the 3 seconds did it. This does give me red flashing and green leds.

---

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco
```

`Linux patrick-HP-ZBook-15u-G5 5.0.0-38-generic #41-Ubuntu SMP Tue Dec 3 00:27:35 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

lf94, 2020-02-17:
@patvdleer now if you follow the rest of what I did, you should be back to square one. :slightly_smiling_face:

conorpp, 2020-02-22:
Closing since I believe everyone's issues have been resolved.

closed 2020-02-22 by conorpp