solokeys/solo1 #367: Soft U2F

Zjemm, 2020-02-01:
Hi guys,

I was just wondering About u2f options when I was playing around with the solo key.

Would it be possible to create a soft u2f solution?

I know krypton is doing something like that using the mobile as the hardware driving the software

Because as nice as a hardware key is... you still need to plug it in

There is also a project from github called softu2f but that’s for macos and there the key is on the same device as where you are logging into

I think this could be a nice next step for solo

jolo1581, 2020-02-01:
Sounds quiete insecure for me.
It's like storing the security key and the database of keepass on the same device.

On my PC I have to use my solo only once until I will logout or my PC hasn't been stored as trusted PC on the website.

One benefit of a hardware token is, that you can carry it with you and you can also use it on untrusted devices. Your password may be stolen, but without security key it isn't possible to login.

Zjemm, 2020-02-01:
That’s where mobile and push comes in like with krypton

jolo1581, 2020-02-02:
Well Android can also turn your mobile into a hardwaretoken.

But I think it isn't a good idea to store the key on the device, which also does the login procedure

Zjemm, 2020-02-02:
What if it would be in the Secure Enclave like on iOS

jolo1581, 2020-02-03:
That might work. 
But does your PC or other device hast something like that? Not everyone has an iOS device.

nekocentral, 2020-02-04:
Android has something like that, as an example.

> Supported devices running Android 9 (API level 28) or higher installed can have a StrongBox Keymaster, an implementation of the Keymaster HAL that resides in a hardware security module. The module contains the following:
> 
> Its own CPU.
> Secure storage.
> A true random-number generator.
> Additional mechanisms to resist package tampering and unauthorized sideloading of apps.
> When checking keys stored in the StrongBox Keymaster, the system corroborates a key's integrity with the Trusted Execution Environment (TEE).
> 
> To support low-power StrongBox implementations, a subset of algorithms and key sizes are supported:
> 
> RSA 2048
> AES 128 and 256
> ECDSA P-256
> HMAC-SHA256 (supports key sizes between 8 bytes and 64 bytes, inclusive)
> Triple DES 168

It can be used for the following categories:

> cryptography:` authorized key algorithm, operations or purposes (encrypt, decrypt, sign, verify), padding schemes, block modes, digests with which the key can be used;
> temporal validity interval: interval of time during which the key is authorized for use;
> user authentication: the key can only be used if the user has been authenticated recently enough. See Requiring User Authentication For Key Use.

Current macs also have the T2 security chip which runs the Secure enclave and on PC you also have Intel SGX and AMD TrustZone. They effectively implements secure enclave, by preventing code running in the regular mode from inspecting or controlling the code running in the trusted execution mode. In the case of Intel, the secure enclave also runs x86 code, while AMD embeds an ARM TrustZone co-processor that only runs the secure enclave program.

Zjemm, 2020-02-05:
So it is possible 👍

nekocentral, 2020-02-05:
It *Might* be possible, but the code base for that would end up so big because all of the cases that it wouldn't be worth it in my opinion

jolo1581, 2020-02-05:
> PC you also have Intel SGX and AMD TrustZone

And all the hardwarebugs in Intel processors are not real 😁

My1, 2020-04-27:
erm question: if you dont want krypton for whatever reason and you dont want softu2f because it's on the same pc, what exactly DO you want.

also if plugging in a solo everytime you wanna log in is too tediious, consider a Somu.
plug in once, keep it there

GalaxyAllie, 2020-06-09:
Honestly, the correct way to implement this is with WebAuthn platform authenticators such as Windows Hello (which is backed by the TPM), Google Chrome's Touch ID support on macOS, Android's ability to use the lock screen as a security key, etc. All of these have secure hardware backing.

This does require website support, but that is growing (for example, they work with GitHub).

My1, 2020-06-10:
well there are a few problems with that

1) TPMs arent everywhere
2) touchid isnt universal on mac (it's only on the Laptops iirc
3) not all OSes have ways for that (like older Windowses like 8.1 which is still supported till 2023 or just Linux in general)
4) for some crazy reason TPMs apparently need their own attestation format (which is dumb, and makes library support a pain)
5) for another crazy reason apparently Win Hello runs on RSA, which makes it even more "fun" to support considering even yubi dropped RSA by now.