solokeys/solo1 #377: Key will not be detected and always starts in bootloader mode

Hannnsen, 2020-02-16:
I was using my Solo successfully on both Windows and Linux. Now since one day it won't get recognized neither by the python interface nor by Firefox. If I plug the key into the USB-port it also flashes his light, like it would be in Bootloader Mode but I have not pressed the button on the key while plugging it in.
Does anybody else has the same problem or could help me out? I would like to use the key again.

conorpp, 2020-02-16:
On Linux you may need to install udev rules to access without root permissions.  If it's in bootloader mode, try updating the firmware.

```
solo key update
```

Hannnsen, 2020-02-16:
Thank you for your reply.
I used it with Linux (Manjaro 18.1.5) and also with Windows 10. In Linux udev is configured as it should be (and it worked the last time without any issues). In Windows 10 drivers are installed and it also worked the last time without any issues. But now it won't get detected by the command line tool (`solo key update` will just say, that it hasn't detected anything). Also Firefox does not detect the key i.e. for 2FA on GitHub.
The weird thing, as I explained: It's blinking like it's in bootloader mode and I don't know how to make a "reset". Any ideas?

lf94, 2020-02-17:
@Hannnsen upload a copy of the official firmware. See my post here: https://github.com/solokeys/solo/issues/364#issuecomment-586665184

Hannnsen, 2020-02-19:
I know tried it with WSL (and configured udev) - because I'm abroad for research I cannot try it on my Manjaro system.

If I do ```solo ls``` I get

```
Traceback (most recent call last):
  File "/home/johannes/.local/bin/solo", line 11, in <module>
    sys.exit(solo_cli())
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/johannes/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/johannes/.local/lib/python3.6/site-packages/solo/cli/__init__.py", line 147, in ls
    solos = solo.client.find_all()
  File "/home/johannes/.local/lib/python3.6/site-packages/solo/client.py", line 56, in find_all
    hid_devices = list(CtapHidDevice.list_devices())
  File "/home/johannes/.local/lib/python3.6/site-packages/fido2/hid.py", line 135, in list_devices
    for d in hidtransport.hid.Enumerate():
  File "/home/johannes/.local/lib/python3.6/site-packages/fido2/_pyu2f/linux.py", line 183, in Enumerate
    for hidraw in os.listdir('/sys/class/hidraw'):
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/hidraw'
```

If I do ```solo key update``` I get
```
Unhandled error connecting to key.
Please report via https://github.com/solokeys/solo-python/issues/
```

Any clue what I should do now?

lf94, 2020-02-20:
@Hannnsen I think the issue is pretty clear: `/sys/class/hidraw` is not available on your system. I guess WSL does not offer a sufficient Linux environment (yet?) to support this. It should work fine on Manjaro.

jolo1581, 2020-02-20:
Missing udev rules?

conorpp, 2020-02-22:
On Windows, you need to run solo tool as admin for it to see security keys.

Hannnsen, 2020-02-23:
> 
> 
> On Windows, you need to run solo tool as admin for it to see security keys.

Thank you for this info to run it as an admin. Now it updated the firmware and the key is running like it should again.

closed 2020-02-23 by Hannnsen