solokeys/solo1 #378: solo key verify error

patvdleer, 2020-02-17:
while trying to setup FIDO via the links below the following happened.
* https://docs.solokeys.io/solo/customization/
* https://docs.solokeys.io/solo/programming#procedure

```
patrick@patrick-HP-ZBook-15u-G5:~/git/solo$ solo program bootloader ~/Downloads/firmware-3.1.1.hex
Not using FIDO2 interface.
erasing firmware...
updated firmware 100%             
time: 7.43 s
patrick@patrick-HP-ZBook-15u-G5:~/git/solo$ solo key verify
Please press the button on your Solo key
Register valid
Solo Hacker firmware.
```
---
```
patrick@patrick-HP-ZBook-15u-G5:~/Dropbox/backups2/somu$ solo program aux enter-bootloader
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.
patrick@patrick-HP-ZBook-15u-G5:~/Dropbox/backups2/somu$ solo program aux enter-dfu

Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)
patrick@patrick-HP-ZBook-15u-G5:~/Dropbox/backups2/somu$ 
patrick@patrick-HP-ZBook-15u-G5:~/Dropbox/backups2/somu$ solo program dfu bundle.hex 
erasing...
downloading 100.00%  08038800 - 08038831 ...         
time: 6208 ms
verifying...
reading 100.00%  08038800 - 08038831 ...         
firmware readback verified.
patrick@patrick-HP-ZBook-15u-G5:~/Dropbox/backups2/somu$ solo program aux leave-dfu
Rewriting option bytes...
Warning: OSError with write_page
Please powercycle the device (pull out, plug in again)
```
---
```
patrick@patrick-HP-ZBook-15u-G5:~/$ solo program bootloader ~/Downloads/firmware-3.1.1.hex
erasing firmware...
Not in bootloader mode.  Attempting to switch...
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.
erasing firmware...
updated firmware 100%             
time: 7.37 s
patrick@patrick-HP-ZBook-15u-G5:~/$ solo key verify
Please press the button on your Solo key
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/solo/client.py", line 237, in make_credential
    attest.verify(data.hash)
AttributeError: 'AttestationObject' object has no attribute 'verify'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/solo", line 10, in <module>
    sys.exit(solo_cli())
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/solo/cli/key.py", line 322, in verify
    cert = solo.client.find(serial, udp=udp).make_credential(pin=pin)
  File "/usr/local/lib/python3.7/dist-packages/solo/client.py", line 240, in make_credential
    verifier().verify(attest.att_statement, attest.auth_data, data.hash)
  File "/usr/local/lib/python3.7/dist-packages/fido2/attestation.py", line 243, in verify
    _validate_packed_cert(cert, auth_data.credential_data.aaguid)
  File "/usr/local/lib/python3.7/dist-packages/fido2/attestation.py", line 214, in _validate_packed_cert
    raise InvalidData('Subject must have OU = "Authenticator Attestation"!')
fido2.attestation.InvalidData: Subject must have OU = "Authenticator Attestation"!
```

patvdleer, 2020-02-17:
FYI I did the following to try and setup the FIDO.

That is the `bundle.hex` from `solo program dfu bundle.hex` above I uploaded.

```sh
git clone --recurse-submodules https://github.com/solokeys/solo.git
cd solo
VERSION_TO_BUILD=3.1.1
git checkout ${VERSION_TO_BUILD}
git submodule update --init --recursive

cd targets/stm32l432
make cbor
make build-hacker
cd ../..

solo key rng raw > seed.bin

curve=prime256v1

country=NL
state=XXX
city=XXX
organization=XXX
unit="Root CA"
CN="Patrick XXX"
email=pat.vdleer@XXX.com

# generate EC private key
openssl ecparam -genkey -name "$curve" -out root_key.pem -rand seed.bin

# generate a "signing request"
openssl req -new -key root_key.pem -out root_key.pem.csr  -subj "/C=$country/ST=$state/O=$organization/OU=$unit/CN=$CN/emailAddress=$email"

# self sign the request
openssl x509 -trustout -req -days 18250  -in root_key.pem.csr -signkey root_key.pem -out root_cert.pem -sha256

# convert to smaller size format DER
openssl  x509 -in root_cert.pem -outform der -out root_cert.der

# print out information and verify
openssl x509 -in root_cert.pem -text -noout

# generate EC private key
openssl ecparam -genkey -name "$curve" -out device_key.pem -rand seed.bin

# generate a "signing request"
openssl req -new -key device_key.pem -out device_key.pem.csr -subj "/C=$country/ST=$state/O=$organization/OU=$unit/CN=$CN/emailAddress=$email"

# sign the request
openssl x509 -req -days 18250  -in device_key.pem.csr -extfile v3.ext -CA root_cert.pem -CAkey root_key.pem -set_serial 01 -out device_cert.pem -sha256

# convert to smaller size format DER
openssl  x509 -in device_cert.pem  -outform der -out device_cert.der

# Verify the device certificate details
openssl x509 -in device_cert.pem -text -noout

echo 'challenge $RANDOM' > chal.txt

# check that they are valid key pairs
openssl dgst -sha256 -sign device_key.pem -out sig.txt chal.txt
openssl dgst -sha256 -verify  <(openssl x509 -in device_cert.pem  -pubkey -noout)  -signature sig.txt chal.txt

openssl dgst -sha256 -sign "root_key.pem" -out sig.txt chal.txt
openssl dgst -sha256 -verify  <(openssl x509 -in root_cert.pem  -pubkey -noout)  -signature sig.txt chal.txt

# Check they are a chain
openssl verify -verbose -CAfile "root_cert.pem" "device_cert.pem"

python3 print_x_y.py device_key.pem

# solo mergehex --attestation-key "b26443XXXXXXXXXXXXXXXXXXXXXXXXXXXX" --attestation-cert device_cert.der --lock solo.hex bootloader.hex bundle.hex

lf94, 2020-02-17:
Unplug and replug? :slightly_smiling_face: 

Also be extremely careful...I see you are messing around in DFU mode. There's really no reason to be in this mode unless you are locking the device. Also `make build-hacker` is deprecated, it should be `make firmware`. I only know any of this because I've spent the past 2 nights sifting through the `solo` repo and history :stuck_out_tongue:

patvdleer, 2020-02-18:
I am far from an expert in FIDO2 but I need to generate my own cert and set it to the somu right? 
Any up to date guide on how to set this up would be awesome.

conorpp, 2020-02-18:
The problem is with this error:

```
fido2.attestation.InvalidData: Subject must have OU = "Authenticator Attestation"!
```

In your log, you changed the OU: `unit="Root CA"`. 

In docs: 

> Note you must use a prime256v1 curve for this step, and you must leave the unit/OU as "Authenticator Attestation".

It is easy to miss, probably should make it more clear in the script itself.

patvdleer, 2020-02-18:
Ok so `solo program bootloader` doesn't affect the bundle.hex I updated via DFU? (sorry new to this)

```
$ solo program bootloader ~/Downloads/firmware-3.1.1.hex
Not using FIDO2 interface.
erasing firmware...
updated firmware 100%             
time: 7.43 s
```

patvdleer, 2020-02-18:
ok so I updated the cert locally should I now run below?

```sh
solo mergehex --attestation-key  97dbXXXXXXXXXXXXXXX  --attestation-cert device_cert.der --lock solo.hex bootloader.hex bundle.hex

solo program aux enter-bootloader
solo program aux enter-dfu
solo program dfu bundle.hex
solo program aux leave-dfu
```

conorpp, 2020-02-18:
`solo program bootloader <firmware.hex>` only updates the application code, and doesn't change any "data".

Looks correct to me.  Note using that `--lock` flag prevents DFU from being access again on the device.

patvdleer, 2020-02-18:
I used lock before, is there any way to override it?

conorpp, 2020-02-18:
No, designed to be permanent :(

patvdleer, 2020-02-18:
Ok so I'm stuck with the `fido2.attestation.InvalidData: Subject must have OU = "Authenticator Attestation"!`

patvdleer, 2020-02-18:
This may be nitpicking but could you move the `If you use --lock, this will permanently lock the device to this new bootloader` notice to above the command...

conorpp, 2020-02-18:
Done!

patvdleer, 2020-02-18:
Ok so is there any use left for this somu?

conorpp, 2020-02-18:
It still looks good in a collection :). Check out @.roycewilliams's impressive stash:
https://twitter.com/TychoTithonus/status/1225875443011637248

[image]

In all seriousness though, I think U2F still works fine if your use case can be configured to only use U2F.  If you're planning to just use it "normally" with various websites, I'm not sure what the compatibility would be like (if this is the case, I'm not sure the benefit to changing the attestation chain anyways).

patvdleer, 2020-02-18:
Awesome collection, I have a graveyard like that somewhere too although not that many auth tokens. 

---

I have multiple somu's I was planning on generating a cert for FIDO and applying it to all 3 having one in the safe, one in the desktop and one in the laptop.

conorpp, 2020-02-18:
If there are other issues, feel free to reopen.

closed 2020-02-18 by conorpp