solokeys/solo1 #380: stm32 libs update

merlokk, 2020-02-19:
looks like we will need to update stm32 libs
gcc-arm-none-eabi-9-2019-q4-major shows this warning

```
                 from src/device.c:8:
src/cmsis/cmsis_gcc.h: In function 'boot_st_bootloader':
src/cmsis/core_cm4.h:93:28: warning: listing the stack pointer register 'sp' in a clobber list is deprecated [-Wdeprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
src/cmsis/cmsis_gcc.h:190:3: note: in expansion of macro '__ASM'
  190 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~

src/cmsis/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
src/cmsis/cmsis_gcc.h:190:3: note: in expansion of macro '__ASM'
  190 |   __ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");
      |   ^~~~~
```

jolo1581, 2020-02-21:
I also see these warnings. After replacing some libs from acutal STM32CubeL4 (release V1.15.0) the warnings are removed.

jolo1581, 2020-03-01:
I did a pullrequest. See: #390