solokeys/solo1 #381: Locked old Solo Hacker

int3l, 2020-02-22:
Hi, I am a kickstarter supporter.
Recently I managed to update my Solo Hacker to version 3.1.1.
I used the command `solo key update --hacker` as per documentation.
But now my solo version is `3.1.1. locked`
My Solo Bootloader is version 2.3.0 still.

Can you please assist me with unlocking my solo firmware please.
I want to have the latest Solo Hacker version and not the Solo Secure version.
Is there a way that I can reprogram or replace the Solo STM32 controller to the initial state with the help of ST-LINK debugger or something like that?

I think that my Solo was locked via web update but my intention was to upgrade to the latest hacker firmware and bootloader - the documentation didn't mention that there will be an implicit lock via the web update.

Please let me know if I can return the key to you or if you can send me unlocked version of the bootloader + firmware (bundle) version.
Thanks in advance!

conorpp, 2020-02-22:
If it shows `locked`, then you cannot use a debugger with it or use the DFU process.  If your bootloader is non-verifying (like on Solo hacker builds), then you can still program non-signed updates.

`solo key update` won't cause a key to lock with recent `3.0.0`+ firmware, but users could mistakenly program with a "secure" version before `3.0.0` to cause the lock setting.

For support issues, please contact [SoloKeys](https://solokeys.com/pages/contact-us) (hello@solokey.com).

closed 2020-02-22 by conorpp

int3l, 2020-02-22:
Oh, sorry. I was instructed via email to post an issue here. I will try hello@.