solokeys/solo1 #383: Option to remove PIN

riedel, 2020-02-23:
It seems that one may only change but never remove a PIN.  As some utils don't support PINs it would be quite useful (otherwise one would have to reset the whole key I guess)

Or did I miss something?

riedel, 2020-02-23:
see also #349

jolo1581, 2020-02-23:
I think you can only clear pin by using solo key reset command. And this will remove also all stored keys.

nickray, 2020-02-23:
I think the reasoning why such an operation is not part of FIDO2 here is that for a resident key (not used as 2FA) you'd always want a PIN, otherwise the RK would just be a single factor (something you have, the key).

riedel, 2020-02-23:
Even with the PIN enabled I can actually generate assertions on a RK that do not need the PIN.

Removing the PIN would need a PIN. There seems no more risk than changing the PIN to eg. 0000.

Tumeez, 2020-03-16:
In my opinion would be great if you can remove your PIN, but it needs to know old PIN then.

Example.

solo key removepin *your pin*
Are you sure that you want remove PIN? (Y/N)
Y
PIN removed successful!

nickray, 2020-03-17:
We have to be standards-compliant here I'm afraid.

nickray, 2020-03-17:
> Even with the PIN enabled I can actually generate assertions on a RK that do not need the PIN.

I believe this should only work if you previously generated an assertion with PIN with the same client (browser).

nickray, 2020-03-17:
OK :) I think we can do this!

- MC says: If pinAuth parameter is not present and clientPin been set on the authenticator, return CTAP2_ERR_PIN_REQUIRED error.
- GA says: If pinAuth parameter is not present and clientPin has been set on the authenticator, set the "uv" bit to 0 in the response.

The only possible dowside is that clients (OS/browser) don't expect removed PIN, which may lead to some confusion when you use a key where the RP expects user verification. But given `remove PIN` would only be possible via CLI, I think it's a better choice to enable "user freedom" here.

Thoughts?

aseigler, 2020-03-18:
I am OK with it as long as:

1. The same anti-hammering technique is used in the remove pin operation as the pin challenge operation (cannot have bad actor physically acquiring token and easily clearing PIN and setting new instead of entering it correctly for UV). Someone might be have a better reasoning for this angle?

2. User is warned that this might cause weird "unknown errors" in the case of visiting a previously registered UV requiring relying party. AFAIK nobody else supports this kind of operation, adding PIN is expected to be one way path. Obviously, this could result in bad UX.