solokeys/solo1 #385: Strategies for unblocking solo on Window 10 >1903

riedel, 2020-02-25:
Is there any ideas for unblocking the solo on Windows 10? 

Windows seems to block all CTAP communication in order to pipe it through windows hello (I guess so that others have the trouble implementing their APIs instead of them needing to push things into standards: smart move, very user friendly way to do corporate fights...)

One idea is using a UsbHID relay as administrator (which seems to be still able to get HID access).

Another Idea would be to open a composite device that exposes exactly the same HID interface under another identifier (is that possible)

My guess is that the alternative API offered by microsoft does not allow many things.

I currently want to use the HMAC-secret extension to get a symetric pass....

aseigler, 2020-02-25:
This is the alternative API: https://github.com/microsoft/webauthn/blob/master/webauthn.h

This is what browsers are using.

My1, 2020-03-27:
yeah w10 is a bit annoying with that.

while it isnt entirely unreasonable to have the OS guard the fido device as some kinda "trusted place", the trust I give MS is borderline zero, and when it blocks functionality it gets annoying, but yeah as far as I know the only way of having it on later w10 is admin.