solokeys/solo1 #389: Can't update to firmware 3.1.2

lpapier, 2020-02-28:
I can't update my solo key (non hacker) from firmware 3.1.1 to 3.1.2 with solo-python (0.0.23 et 0.0.24).

```bash
$ solo key update
Wrote temporary copy of firmware-3.1.2.json to /tmp/tmptp852iv4.json
sha256sums coincide: 4a24fae228b5c5560d5ed8982770a72dec3cae5dcb02f6fc768b03a60afb10cf
Switching into bootloader mode...
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 10.72 s
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters
```

The key is left in bootloader mode.

Flashing from bootloader mode also failed.

```bash
$ solo program bootloader firmware-3.1.2.json
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.41 s
Traceback (most recent call last):
  File "/home/papier/bin/solo", line 10, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3.7/site-packages/click/core.py", line 763, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 716, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 1136, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.7/site-packages/click/core.py", line 1136, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.7/site-packages/click/core.py", line 955, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 554, in invoke
    return callback(*args, **kwargs)
  File "/home/papier/.local/lib/python3.7/site-packages/solo/cli/program.py", line 173, in bootloader
    raise e
  File "/home/papier/.local/lib/python3.7/site-packages/solo/cli/program.py", line 168, in bootloader
    p.program_file(firmware)
  File "/home/papier/.local/lib/python3.7/site-packages/solo/client.py", line 402, in program_file
    self.verify_flash(sig)
  File "/home/papier/.local/lib/python3.7/site-packages/solo/client.py", line 211, in verify_flash
    self.exchange(SoloBootloader.done, 0, sig)
  File "/home/papier/.local/lib/python3.7/site-packages/solo/client.py", line 151, in exchange_hid
    raise CtapError(ret)
fido2.ctap.CtapError: CTAP error: 0x27 - OPERATION_DENIED
```

Re-flashing firmware 3.1.1 works

```bash
$ solo program bootloader firmware-3.1.1.json
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.40 s
$ solo key verify
Please press the button on your Solo key
Register valid
Valid Solo with firmware from SoloKeys.
```

conorpp, 2020-02-28:
Can you share the bootloader version?

When key is in bootloader / update mode:
```
solo key version
```

lpapier, 2020-02-28:
```bash
$ solo key version
Not using FIDO2 interface.
3.0.0 unlocked

conorpp, 2020-02-28:
Did you program this key yourself, or buy it as "secure"?

lpapier, 2020-02-28:
I buy it as a "secure" and update it from firmware 3.1.0 to 3.1.1 with solo-python 0.0.23 without any problem.

conorpp, 2020-02-28:
Updating to 3.1.1 (again) with solo-python 0.0.24 works fine?

lpapier, 2020-02-28:
I'm sorry. How can I update my key to 3.1.1 again ? My key is already in 3.1.1.

```bash
$ solo key version
3.1.1 locked
$ solo key verify
Please press the button on your Solo key
Register valid
Valid Solo with firmware from SoloKeys.
```

conorpp, 2020-02-28:
Just run `solo program bootloader firmware-3.1.1.json` like you did before, but with solo-python 0.0.24.

lpapier, 2020-02-28:
Works fine!

```bash
$ solo version
0.0.24
$ solo key version
3.1.1 locked
$ solo program bootloader firmware-3.1.1.json
using signature version <=2.5.3
erasing firmware...
Not in bootloader mode.  Attempting to switch...
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%             
time: 7.41 s
```

conorpp, 2020-02-28:
Did you try programming `3.1.2` multiple times with same error?

lpapier, 2020-02-28:
Yes, several time with solo-python 0.0.23 and 0.0.24 on 2 different computers.

schwarzeh, 2020-02-29:
I have the same problem...

~~~
D:\> solo key update
Wrote temporary copy of firmware-3.1.2.json to F:\Temp\System\tmptsxgzo0q.json
sha256sums coincide: 4a24fae228b5c5560d5ed8982770a72dec3cae5dcb02f6fc768b03a60afb10cf
Switching into bootloader mode...
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 10.75 s
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters

D:\> solo key version
Not using FIDO2 interface.
3.0.0 unlocked

D:\> solo program bootloader firmware-3.1.1.json
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.36 s

D:\> solo key version
3.1.1 locked

D:\> solo program bootloader firmware-3.1.2.json

using signature version <=2.5.3
erasing firmware...
Not in bootloader mode.  Attempting to switch...
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.37 s
Traceback (most recent call last):
  File "d:\python\lib\site-packages\solo\cli\program.py", line 168, in bootloader
    p.program_file(firmware)
  File "d:\python\lib\site-packages\solo\client.py", line 397, in program_file
    self.write_flash(i, data)
  File "d:\python\lib\site-packages\solo\client.py", line 199, in write_flash
    self.exchange(SoloBootloader.write, addr, data)
  File "d:\python\lib\site-packages\solo\client.py", line 147, in exchange_hid
    data = self.send_data_hid(SoloBootloader.HIDCommandBoot, req)
  File "d:\python\lib\site-packages\solo\client.py", line 142, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "d:\python\lib\site-packages\fido2\hid.py", line 94, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "d:\python\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "d:\python\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\python\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "d:\python\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "d:\python\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "d:\python\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\python\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\python\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\python\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "d:\python\lib\site-packages\solo\cli\program.py", line 184, in bootloader
    p.program_file(firmware)
  File "d:\python\lib\site-packages\solo\client.py", line 409, in program_file
    self.verify_flash(sig)
  File "d:\python\lib\site-packages\solo\client.py", line 211, in verify_flash
    self.exchange(SoloBootloader.done, 0, sig)
  File "d:\python\lib\site-packages\solo\client.py", line 151, in exchange_hid
    raise CtapError(ret)
fido2.ctap.CtapError: CTAP error: 0x27 - OPERATION_DENIED

D:\> solo key version
Not using FIDO2 interface.
3.0.0 unlocked

D:\> solo program bootloader firmware-3.1.1.json
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.37 s
~~~

conorpp, 2020-02-29:
I'm having trouble reproducing since none of my 3.0.0 bootloader keys fail on the update.  I'd really like to solve this issue though.  Can one (or both) of you mail your key to me to troubleshoot?  If so, please send an email to hello@solokeys.com to coordinate.  Will send you a replacement key of course.

ncth, 2020-03-01:
Same issue here, 3.1.1 works fine but as soon as I try to upgrade to 3.1.2 it fails.

Picostar21, 2020-03-04:
I have the same problem, it goes in the bootloader and then i have to set it back to 3.1.1

pkubin, 2020-03-04:
Same issue here. I got a secure USB-A version with bootloader 3.0.0. The signature verification fails and i have to upload the 3.1.1 firmware afterwards. I had no problem updating my older secure USB-A+NFC stick with a 2.5.3 bootloader.

schwarzeh, 2020-03-04:
My somu is on the way to Connor (from Germany to the USA). Not much longer, and the problem should be solved :)

lpapier, 2020-03-06:
@conorpp I sent you an email on Saturday... I guess google tag it as spam. If you still need another key feel free to ask for.

@schwarzeh Nice to hear that one key is flying to USA.

NathanielJS1541, 2020-03-06:
I'm also getting the same issue - I attempted to go from `3.1.1` to `3.1.2` and got the same error:

```
PS C:\WINDOWS\system32> solo key update
Not using FIDO2 interface.
Wrote temporary copy of firmware-3.1.2.json to C:\Users\Natha\AppData\Local\Temp\tmpdpgwfuwz.json
sha256sums coincide: 4a24fae228b5c5560d5ed8982770a72dec3cae5dcb02f6fc768b03a60afb10cf
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.36 s
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters
```

NathanielJS1541, 2020-03-06:
I'm not sure if this is useful or not, but I get a different error when trying to update using the released firmware-3.1.2.json file:

This is when the key is not already in bootloader mode:
```
PS C:\Users\Natha\Downloads> solo program bootloader firmware-3.1.2.json
using signature version <=2.5.3
erasing firmware...
Not in bootloader mode.  Attempting to switch...
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.30 s
Traceback (most recent call last):
  File "c:\program files\python38\lib\site-packages\solo\cli\program.py", line 168, in bootloader
    p.program_file(firmware)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 397, in program_file
    self.write_flash(i, data)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 199, in write_flash
    self.exchange(SoloBootloader.write, addr, data)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 147, in exchange_hid
    data = self.send_data_hid(SoloBootloader.HIDCommandBoot, req)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 142, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "c:\program files\python38\lib\site-packages\fido2\hid.py", line 94, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\program files\python38\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\program files\python38\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python38\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\program files\python38\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\program files\python38\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\program files\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\program files\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\program files\python38\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\program files\python38\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\program files\python38\lib\site-packages\solo\cli\program.py", line 184, in bootloader
    p.program_file(firmware)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 409, in program_file
    self.verify_flash(sig)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 211, in verify_flash
    self.exchange(SoloBootloader.done, 0, sig)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 151, in exchange_hid
    raise CtapError(ret)
fido2.ctap.CtapError: CTAP error: 0x27 - OPERATION_DENIED
```

And this is when already in bootloader mode:
```
PS C:\Users\Natha\Downloads> solo program bootloader firmware-3.1.2.json
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 7.30 s
Traceback (most recent call last):
  File "c:\program files\python38\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\program files\python38\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python38\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\program files\python38\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\program files\python38\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\program files\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\program files\python38\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\program files\python38\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\program files\python38\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\program files\python38\lib\site-packages\solo\cli\program.py", line 173, in bootloader
    raise e
  File "c:\program files\python38\lib\site-packages\solo\cli\program.py", line 168, in bootloader
    p.program_file(firmware)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 409, in program_file
    self.verify_flash(sig)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 211, in verify_flash
    self.exchange(SoloBootloader.done, 0, sig)
  File "c:\program files\python38\lib\site-packages\solo\client.py", line 151, in exchange_hid
    raise CtapError(ret)
fido2.ctap.CtapError: CTAP error: 0x27 - OPERATION_DENIED
```

Not sure if this helps but I thought it may be worth adding.

clar-cmp, 2020-03-06:
Same here-- updating to 3.1.2 fails.

woakas, 2020-03-06:
Here with this error:

➜ ~ solo key update --secure
We are about to update with the latest Solo Secure firmware.
Please confirm that the connected Solo key is a Solo Secure [y/N]: y
Could not fetch official firmware build from solokeys/solo repository
releases!
URL attempted:
https://github.com/solokeys/solo/releases/download/3.1.2/firmware-secure-3.1.2.json

NicoPy, 2020-03-07:
I have the same problem with 1 SOMU hacker key converted to locked mode.
I don't have the problem with 2 other SOMU hacker keys converted to locked mode.
Also, I don't have the problem with 2 SOLO keys (non hacker).

dmindfuse, 2020-03-08:
Same here... anyone know how to solve it yet? The issue is as described above.

Somebodyisnobody, 2020-03-09:
I have a similar problem: After a failed flashing with Solo Desktop the LED is pulsing red/green. How can I force downgrade with Solo Desktop?

JGReyes, 2020-03-12:
I also have the same problem with 2 Solo keys(non hacker). Same tests and output than lpapier.

conorpp, 2020-03-16:
I figured out the error with 3.1.2, and should [be fixed now with 3.1.3.](https://github.com/solokeys/solo/pull/396)  `solo key update` should now work.  Please reopen if there's still an issue.

closed 2020-03-16 by conorpp