solokeys/solo1 #394: Solo Hacker bricked trying to update bundle-hacker-XXX.hex

matthewmcneely, 2020-03-11:
I built a new bundle (with bootloader):

```
+ /opt/conda/bin/solo mergehex --attestation-key XXXXXXX --attestation-cert /solo/keys/device_cert.der bootloader-nonverifying-3.0.1-30-gbcbc65e.hex firmware-3.0.1-30-gbcbc65e.hex bundle-hacker-3.0.1-30-gbcbc65e.hex
```

Followed directions to write this bundle:

```
$ solo program aux enter-bootloader
Not using FIDO2 interface.
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.

$ solo program aux enter-dfu
Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)
```

Powercycled, then:

```
$ solo program dfu builds/bundle-hacker-3.0.1-30-gbcbc65e.hex
Traceback (most recent call last):
  File "/Users/me/code/solo/venv/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/solo/cli/program.py", line 63, in dfu
    dfu = solo.dfu.find(serial, attempts=connect_attempts)
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/solo/dfu.py", line 30, in find
    dfu.find(ser=dfu_serial, dev=raw_device, altsetting=altsetting)
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/solo/dfu.py", line 98, in find
    usb.core.find(idVendor=0x0483, idProduct=0xDF11, find_all=True)
  File "/Users/me/code/solo/venv/lib/python3.7/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

Yikes, was hoping that was just a warning. But no, the Solo is bricked.

```solo program aux leave-dfu``` or following the button press recovery technique offers no help.

conorpp, 2020-03-11:
Looks like the first command could not find libusb to be able to talk to the device in DFU mode.  Very likely the device is in DFU mode still and not bricked.  Try installing libusb for your system.  What OS are you on?

conorpp, 2020-03-11:
For windows:
https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/1.2.6.0/

matthewmcneely, 2020-03-11:
@conorpp Thanks for your help. 

I'm on OSX. Note that I'm able to update the firmware using the `solo` command, along with `key reset`, etc. So I don't _think_ is a libusb error.

conorpp, 2020-03-12:
The regular update uses USB HID protocol to communicate, and is much better supported on platforms.  However, the DFU is a more custom USB protocol and HID cannot be used, so it has to use a more "low-level" backend to communicate, like libusb.  For Macos, it looks like you can use brew to install it: `brew install libusb` or `brew install libusb-compat`.

matthewmcneely, 2020-03-12:
@conorpp Thanks, that was the issue (needed libusb)

closed 2020-03-12 by matthewmcneely