solokeys/solo1 #403: Bug when getting credentials metadata

rgerganov, 2020-03-26:
If we try to get credential metadata and there are no resident keys on the device, it returns `FIDO_ERR_NO_CREDENTIALS`. This is not the correct behavior. We should return 0 existing and 50 remaining keys.

Will submit a patch shortly. /cc @conorpp

rgerganov, 2020-03-26:
I found this issue when testing with the security manager in Google Chrome and then found 3 more issues, see the second commit in the PR.

Many thanks to @My1 for providing debug logs from a key that works with Chrome :)

My1, 2020-03-26:
no problem, I wanna help solo too, also to show how helpful it is to have updates on a key. (maybe even the ppl with proprietary keys will notice someday that it's useful to not have them on full lockdown), lol

closed 2020-03-28 by conorpp