solokeys/solo1 #411: SoloKeys constantly blinking red

emafh, 2020-03-29:
I recently bought a SoloKeys, at the beginning it worked good. But now, when I insert it, it is constantly blinking and not reacting at all anymore. Any solution?

jolo1581, 2020-03-29:
What about using solo Python?
Do you get an output if you use `solo ls`?

My1, 2020-03-29:
no idea which Solokeys device you bought but chances are high that you entered the bootloader, especially when dealing with a Somu (a tiny Version of the Solo).

Both devices enter the bootloader (for example to do updates) if the key is pressed for a few seconds upon powerup.

try to not press/touch the button for too long when inserting then and in case of the somu try using your finger nails to insert the Somu.

nickray, 2020-04-01:
Another possibility is you initiated a firmware update (e.g. via old webpage) and aborted during the process. Then you'd just have to run another update.

My1, 2020-04-01:
true enough when the bootloader cant start it iirc stays in the BL. I think the webupdater works best in chrome on anything that isnt mobile or Windows 10 past 1903 (so, windows 10 up to 1809, older windowses that support chrome with Webauthn, linux and possibly mac and whatever desktop OSes might float around that run chrome should kinda work)

jolo1581, 2020-04-02:
But if you in BL mode you should see something using `solo ls`

szszszsz, 2020-04-02:
And it should be all colors, not only red, unless something has changed recently.

My1, 2020-04-02:
> And it should be all colors, not only red, unless something has changed recently.

Actually iirc it was only green and red flashing for bl and depending on the case and solo used the colors may not be accurate and bl mode on a red case can look a lot like yellow/orange juat blinking

jolo1581, 2020-04-02:
Since @emafh won't give us more information, we could only guess what the problem is.