solokeys/solo1 #412: Resident Key Login only lists 19 keys

My1, 2020-03-31:
despite having more than 19 keys registered for an RP apparently only the first 19 show up.

conorpp, 2020-03-31:
When running getAssertion, all the "matching" assertions need to be buffered on the first request, so there's a hard ram limit to how many can get buffered.  If I recall correctly, it should be 19 or 20.

My1, 2020-03-31:
okay, I wasnt aware of that as at least some other Fido devices (notably the G310) allows for FAR higher numbers with that

nickray, 2020-04-01:
Technically I think getAssertion just needs to count them and then keep track of "next" index for getNextAssertion calls. It's more efficient to cache them though, and honestly I doubt there is a practical use case for so many RKs for the same RP on on key, no?

My1, 2020-04-01:
I dunno whether there are any practical use cases (except maybe people not being aware credential management is a thing) I just noticed it happen and reported it as it is probably not supposed to happen to not list all RKs.

other solutions might be blocking RK regs on an RP beyond cache limit (although no idea whether there's an error code for "I think that's a bad idea, let's not do this")