solokeys/solo1 #420: multiple solo/somu 4.0.0 with U2F chrome doesn't work in gitlab and nextcloud

guerby, 2020-04-06:
Hi,

Using chrome 80.0.3987.163 (Official Build) (64-bit) on debian 10 we found an issue when registering more than one solo/somu key firmware 4.0.0 as U2F in gitlab (12.9.2-ee.0 omnibus on debian 10) and nextcloud (18.0.3 onubuntu 18.04.4 and VM script install).

With one registered solo key all work fine, when registering a second key chrome starts to refuse U2F login with one of the key.

firefox 74.0.1 doesn't have the issue.

Registering one solo, one yubikey and two nitrokey works, it's only when the second solo is registered that the issue happens.

It seems that we didn't have this issue with firmware 3.0.

I don't know where to look to produce useful logs.

If you don't have access to a gitlab or nextcloud instance to reproduce the issue let me know.

guerby, 2020-04-06:
I just retested to be sure : no issue with two SOMU firmware 3.0.0, I unregister, upgrade to 4.0.0, register both key and then the issue appears = only one somu works to login the other one has "Authentication via U2F device failed.".

guerby, 2020-04-06:
FYI after reading https://github.com/solokeys/solo/issues/415 I did a "solo key reset" on both keys, registered again and the issue is still there so seems not linked to rk.

guerby, 2020-04-20:
I tested registering solo 4.0.0  on meshcentral2 sofware (nodejs web) and no issue there, so it must be something specific to the way gitlab and nextcloud interact with 4.0.0 firmware.

My1, 2020-04-27:
well both gitlab and nextcloud are not running with RKs so no RK issues to be expected

can you maybe try other 3.x versions?

also maybe try my little fun sandbox tool https://my1.dev/wa/my1.php
(with the same userid as that's basically what counts as "one account") whether that's also affected. also maybe play with the parameters a bit when logging in.
(if your Somu/solos have a PIN and you have used it in the sandbox with that credential and want to try discouraged you need to enable the override or the server is gonna kick you, metaphorically)

my little sandbox is basically as easy as a webauthn implementation can get so this should work.

also Maybe I'll try making some sandboxes with U2F (both 1.0 and 1.1 javascript versions as they are a bit different) so there's more fun.

because maybe the stuff behaves differently for Fido2 and U2F...

guerby, 2020-04-28:
> well both gitlab and nextcloud are not running with RKs so no RK issues to be expected
> 
> can you maybe try other 3.x versions?
> 
> also maybe try my little fun sandbox tool https://my1.dev/wa/my1.php
> (with the same userid as that's basically what counts as "one account") whether that's also affected. also maybe play with the parameters a bit when logging in.
> (if your Somu/solos have a PIN and you have used it in the sandbox with that credential and want to try discouraged you need to enable the override or the server is gonna kick you, metaphorically)
> 
> my little sandbox is basically as easy as a webauthn implementation can get so this should work.
> 
> also Maybe I'll try making some sandboxes with U2F (both 1.0 and 1.1 javascript versions as they are a bit different) so there's more fun.
> 
> because maybe the stuff behaves differently for Fido2 and U2F...
> 
> also on which OS are you (on the browser side) in case of windows 10, which version are you on (as 1903 and later treat Fido Stuff VERY differently)

Using debian 10 (buster) with various browsers (chrome/chromium 81 and firefox 74.0.1 currently).

I don't know how to register multiple keys on the same account on https://my1.dev/wa/my1.php

My1, 2020-04-28:
> I don't know how to register multiple keys on the same account on https://my1.dev/wa/my1.php

self quote

> (with the same userid as that's basically what counts as "one account")

just enter the same user ID for all registrations (just make one up you like)

guerby, 2020-04-28:
> > I don't know how to register multiple keys on the same account on https://my1.dev/wa/my1.php
> 
> self quote
> 
> > (with the same userid as that's basically what counts as "one account")
> 
> just enter the same user ID for all registrations (just make one up you like)

I have to use "Force Submit" to login but it seems to work with multiple keys.

My1, 2020-04-28:
> I have to use "Force Submit" to login but it seems to work with multiple keys.

probably because I commented out auto-submit again when I wanted to fiddle with some values, no weird thing. but I just noticed gitlab doesn't spawn any browser popups like webauthn generally does so maybe it's specific to U2F, this is fun...

guerby, 2020-04-28:
Nextcloud and gitlab are both U2F. I tested again with chrome 81 and latest gitlab 12.10.1-ee.0 and I still get an issue when I register two keys, one of the two no longer work for login. If I remove the "working" one, the "not working" one starts to work again.

guerby, 2020-04-28:
> probably because I commented out auto-submit again when I wanted to fiddle with some values, no weird thing. but I just noticed gitlab doesn't spawn any browser popups like webauthn generally does [...]

My1, 2020-04-28:
> Nextcloud and gitlab are both U2F. I tested again with chrome 81 and latest gitlab 12.10.1-ee.0 and I still get an issue when I register two keys, one of the two no longer work for login. If I remove the "working" one, the "not working" one starts to work again.

is it a specific one of the two that isnt working or is it always the first/second to be registered or is it kinda random?

guerby, 2020-04-28:
> is it a specific one of the two that isnt working or is it always the first/second to be registered or is it kinda random?

Second added seem to keep working, first added doesn't work.

Do you have more than one solo/somu to test on nextcloud/gitlab instances ?

My1, 2020-05-02:
Okay I finally took some time to write my U2F sandbox

https://my1.dev/u2f/my1.php

and I really have more insight here.

so I have a Solo-A (non-tap) and a Somu both Secure and on 4.0.0

and for some crazy reason both attempt using the first keyhandle.

so I guess the Solo forgot to put some checks that the keyhandle belongs to that specific one whch is why it answers for one that doesnt belong to it.

this obviously throws a signature Error (and generally services do not expose server errors so the sandbox is the best way.

on the other hand, the registration code does somehow check that otherwise you would get an error that the key already exists. so it's a kinda weird error that only happens upon u2f.sign but not upon u2f.register, and only happens on U2F.

guerby, 2020-05-02:
There's only been two commits by @conorpp since 4.0.0 don't know if they're related to this issue.

@My1 did you try with 3.x firmware ?

My1, 2020-05-02:
I not actually tried intentionally with 3 firmware. But one of the 2 was on 3.1.1 and it iirc was the first registered soni think that might have worked. Can't say for sure tho and i obviously updated when i noticed that.

Regarding the 2 commits I doubt they related to this, as one is a problem specific to resident keys and the other for hmac-secret so both on FIDO2 level.