solokeys/solo1 #421: solo (-python) does not work for a user but for root

SchwarzerA, 2020-04-11:
Hi guys,

although solo-python seems to be installed and okay there is an issue when running it as normal user (as recommended):

root@sirwiz: ;(RC=0) # solo
Processing… 
THIS COMMAND SHOULD NOT BE RUN AS ROOT!

Please install udev rules and run `solo` as regular user (without sudo).
…
shw@sirwiz: ~;(RC=1) $ solo
Processing… 
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 5, in 
    from solo.cli import solo_cli
ModuleNotFoundError: No module named 'solo'

Do you have any hints (the web has not)?

Thank you in advance.

openSUSE TumbleWeed, Python 3.8

conorpp, 2020-04-11:
Maybe try installing not as root?  And did you use pip3?

My1, 2020-04-18:
also if that works but you cant interact with the solos you need udev rules