solokeys/solo1 #425: Not able to use Solo to sign in to Microsoft account

lumaxis, 2020-04-21:
When trying to use the "Sign in with a security key" option for signing in to my Microsoft account with Chrome, I get prompted for the (required) PIN but right after, I run into this error:

[image]

Not sure what other info I should provide for this to be helpful. Or is this rather a bug with the browser?

aseigler, 2020-04-22:
Which build of Windows (Win+R, winver.exe), and Chrome (Help, About), and firmware revision of Solo?  Seen some issues lately with Chrome 81, but not like this report.

jolo1581, 2020-04-22:
You had set a pin to your solo?

`solo keyset-pin`

lumaxis, 2020-04-22:
@aseigler Sorry, should have provided this initially already 😅
This is actually on macOS 10.15.4 and with Chrome 81.0.4044.122. The Solo firmware version is 4.0.0 locked.

@jolo1581 Yes, a PIN is set and I get prompted for it. Entering a wrong PIN doesn't let me proceed, only the correct PIN leads to the above error page.

jolo1581, 2020-04-22:
@lumaxis can you tell us the solo versionnumber. Is it already 4.0.0.

The major version 4 has many updates with resedent Key management. Maybe something ist broken. If you have major 4 maybe you have a solo with older firmware version

lumaxis, 2020-04-22:
Yes, see above 🙂 It’s already 4.0.0

aseigler, 2020-04-22:
> @aseigler Sorry, should have provided this initially already 😅
> This is actually on macOS 10.15.4 and with Chrome 81.0.4044.122. The Solo firmware version is 4.0.0 locked.
>
> @jolo1581 Yes, a PIN is [...]

I just tried with the exact same versions as you and could not reproduce the problem.  I haven't looked at this login flow lately, but something has changed significantly for the better, the assertion data is now plainly visible which makes issues like this FAR easier to debug.  You should be able to hit F12 before you try the login, then look for the "post.srf?wa=wsignin1.0...." entry, and at the bottom under form data, in the assertion, the thing being complained about is the userHandle value.

lumaxis, 2020-04-23:
@aseigler Looks I accidentally lied, I'm trying to log in via https://dev.azure.com which eventually goes to https://go.microsoft.com/fwlink/?LinkId=2014676&githubsi=true but should in theory use the same Microsoft login.
When looking at the request data, the `userHandle` actually looks the same. Not sure what's wrong here …

aseigler, 2020-04-24:
Ah! OK, I just tried dev.azure.com from Windows 10 with TPM rather than Solo and got the exact same message via login.microsoftonline.com, although the assertion looks good and the userHandle looks reasonable.

lumaxis, 2020-07-16:
Seems like this has been fixed on the Microsoft side 🎉

closed 2020-07-16 by lumaxis