solokeys/solo1 #426: Solo Secure Key keeps blinking orange / green

wolfpfannenstiel, 2020-04-22:
My brand new Solo Secure Key does not work. The LED is blinking green / orange alternately as soon as I plug it into the USB port. The Solo Desktop App (on Windows 10) recognizes the key, tells me the serial number (206F37AE414B) and says "device ready". However, trying to update the key to the newest firmware version (from 2.5.3 to 4.0.0.) first shows a success message after an erasing and writing process but eventually results in an error message "Error: CTAP2_ERR_OPERATION_DENIED (0x27)". This error message is of no help to me and unfortunately the documentation on solo keys that can be found in the Web is only rudimentary. (Maybe I haven't found the right place yet.) The firmware version is still shown to be 2.5.3. 
Almost all of the available actions such as "Verify", "Register", "Change PIN", and "Reset" result in a white "1" shown in a red box as their only result. Pressing "Authenticate" tells me to register first, which does not work, as stated before. 
Testing WebAuthn on webauthn.io says it cannot recognize the key. 
Pressing the button for five seconds when plugging the key into the PC does not have any effect. The LED keeps blinking orange and green. 
What is wrong?
Thanks and best regards
Wolf

jolo1581, 2020-04-23:
Can you check your software version using `solo ls` and `solo key verify` with solo python.

wolfpfannenstiel, 2020-04-23:
solo ls
Not using FIDO2 interface.
:: Solos
206F37AE414B: Solo Bootloader 2.5.3

 solo key verify
Please press the button on your Solo key
Not using FIDO2 interface.
Traceback (most recent call last):
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\WolfPfannenstiel\AppData\Local\Programs\Python\Python38\Scripts\solo.exe\__main__.py", line 9, in <module>
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\solo\cli\key.py", line 323, in verify
    cert = solo.client.find(serial, udp=udp).make_credential(pin=pin)
  File "c:\users\wolfpfannenstiel\appdata\local\programs\python\python38\lib\site-packages\solo\client.py", line 238, in make_credential
    attest, data = self.client.make_credential(options, pin=pin)
AttributeError: 'NoneType' object has no attribute 'make_credential'

jolo1581, 2020-04-24:
OK, your stucks in bootloader mode and hast no application software.

Try `solo key update` to flash actual firmware to your solo.
After update verify with:
`solo ls` should show something like that:
[image]

Then try `solo key verify`. The answer should be like that:
[image]

I guess this should solve your problem.

wolfpfannenstiel, 2020-04-24:
Thanks, that worked! The key is now back to normal and I successfully tested it on WebAuthn.io.

However, I must say that the skills needed to fix the issue are probably beyond those of non-IT people. Fortunately, I am an "IT guy" and so I was able to install python, which was not on my machine before, compile the solo tools and execute the commands you proposed. I can't imagine non-IT people (such as e.g. my family members or most of my friends) could do this without expert help. And they surely wouldn't be willing to do this. This technology is apparently not yet ready for plug-and-play. Furthermore, as I stated in my first message, the documentation I was able to find on the web is rather "thin". 

Nevertheless, thanks again for your support. The key is working now and I managed to set it up as second authentication factor for Github.

closed 2020-04-24 by wolfpfannenstiel