solokeys/solo1 #427: Updating bootloader for Solo Key

Iolaum, 2020-04-25:
I have 2 solo keys, 1 secure and 1 hacker.

I just updated to firmware version 4.0.0 but I noticed that this process still leaves me with the old bootloader. Here are some outputs from the command line to show the state of the secure key.

When entering the solo key in normal mode:

```
$ solo ls
:: Solos
207636905548: SoloKeys Solo 4.0.0
```

When entering the solo key in bootloader more:
```
$ solo ls
Not using FIDO2 interface.
:: Solos
207636905548: SoloKeys Solo Bootloader 2.0.0
```

I see that the [releases page](https://github.com/solokeys/solo/releases) also has bootloader binaries listed. I couldn't find relevant instructions in the documentation. Is there a way to also update the solo bootloader for my keys?

Iolaum, 2020-04-25:
Note that while looking for a solution I found [this github post](https://github.com/solokeys/solo/issues/199#issuecomment-593654868).

I tried entering my solokey in bootloader mode and then tried to enter DFU mode, but it didn't work:

```
$ solo ls
Not using FIDO2 interface.
:: Solos
207636905548: SoloKeys Solo Bootloader 2.0.0
$ solo program aux enter-dfu
Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)
```

Plugging out and in the device puts it in either normal or bootlaoder mode depending on whether I press the solo button.

conorpp, 2020-04-25:
For "secured" Solos, the bootloader cannot be updated.  But the hacker models can.  Check the docs here for more information.

https://docs.solokeys.io/solo/programming/

Iolaum, 2020-04-25:
@conorpp Thank you for the clarification.

I was wondering. If I am at DFU mode and want to update the solo hacker key to include the latest version of the bootloader do I run:

```
solo program dfu bootloader-nonverifying-4.0.0.hex
```

or should I use bundle-hacker-4.0.0.hex` running
```
solo program dfu bundle-hacker-4.0.0.hex
```

It's not clear (for me) from the docs what file I should pick. I don't mind updating both the "application" part as well as the bootloader part, looking for the less risky option.

I do want the solo hacker key to retain it's hacker "feature".

conorpp, 2020-04-25:
`bundle-hacker-4.0.0.hex` contains the bootloader, application, and attestation information.  `bootloader-nonverifying-4.0.0.hex` is just the bootloader.  You should use the bundle to get everything you need.  Any DFU update erases everything, so programming `bundle-hacker-4.0.0.hex` would leave you with only a bootloader.

Iolaum, 2020-04-25:
Thanks a lot! This worked!

Two notes:

1. It would be good to specify in the docs that using the DFU mode to write something completely removes existing data/partitions. It is implied from warning 1 but not as clear for newcomers. I can make a PR clarifying that if you want.
2. I needed to use `sudo` for commands to work while in DFU mode. Otherwise I was getting `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`. It's likely due to insufficient udev rules on my machine (Fedora MATE 31) though it's strange that this was only needed in DFU mode and not in normal or bootloader mode.

Will close this issue as key question has been answered.

closed 2020-04-25 by Iolaum

conorpp, 2020-04-25:
Happy to help, and really appreciate any documentation PRs, especially from someone new to the process :)

nickray, 2020-04-26:
> 2. I needed to use `sudo` for commands to work while in DFU mode. Otherwise I was getting `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`. It's likely due to insufficient [...]

This is to be expected. USB devices are distinguished by their VID/PID pairs. Solo and Bootloader Solo are "our" devices, and share the same VID/PID pair. Once you're in bootloader mode, you're running code by STMicroelectronics that's in the chip's ROM and runs under a VID/PID pair of theirs.

Iolaum, 2020-04-26:
:o good to know.

I ended up solving the problem by adding solo udev rules manually

```
$ cd /etc/udev/rules.d
$ sudo wget https://raw.githubusercontent.com/solokeys/solo/master/udev/70-solokeys-access.rules
```

Good job that you have added the STMicroelectronics relevant rules there!