solokeys/solo1 #430: Error building on Windows: No rule to make target 'solo.elf', needed by 'solo.hex'

AdamUCF, 2020-04-29:
I've been unable to compile on Windows both in WSL and natively in Windows. In both cases I get the following error when running "make firmware"...

make[1]: *** No rule to make target 'solo.elf', needed by 'solo.hex'.  Stop.

I'll admit that I'm a bit of a rookie when it comes to building such things but I've followed the [building instructions](https://docs.solokeys.io/solo/building/) (install the ARM compiler, python tools, etc) and this is as far as I can seem to get. Is building on Windows supposed to work? I assumed Windows itself may be a tough hurdle but figured it would at least work in WSL/Ubuntu on Windows but even that isn't working. Any suggestions would be much appreciated.

aseigler, 2020-04-30:
In Debian in WSL with Python3 etc installed:

`git clone --recurse-submodules https://github.com/solokeys/solo`
`cd solo`
`make venv`
`source venv/bin/activate`
`cd targets/stm32l432`
`make cbor`
`make firmware`

AdamUCF, 2020-04-30:
Ahhhh that worked. Thank you!

closed 2020-04-30 by AdamUCF