solokeys/solo1 #431: terminal login - which PAM file?

usrstephan, 2020-05-03:
This question is rather about solokey integration:
I use Ubuntu (20.04) as a user without sudo permissions. Before doing sudo actions in terminal/commandline I have to insert a password two times: the first time to change the user (su [sudouser]) and the second time to allow for sudo actions (sudo apt update).
Following the documentation, the solokey authentication for sudo commands works fine (editing /etc/pam.d/sudo; the second authentication).
Is it possible to do the first authentication with solokey? Which PAM-file do I have to edit?

nickray, 2020-05-03:
I think you should do `sudo -u <user> <your command>` (or `sudo -u <user> -i` etc.)?

usrstephan, 2020-05-04:
thanks for your hint - but that does not work as the normal user has no sudo priviledges.
What I probably need is the PAM file responsible for user authentication in terminal.

jolo1581, 2020-05-04:
I wondered that you use `su user`.
In older Ubuntu versions (e.g. 18.04) `sudo command` works fine without doing the way over root.
With the su command, you work as user root. The behavior you have is not a solo problem. 

Two solutions:
The better way ist to add your user to sudoers. So you can Work with `sudo command`
The bad way is:
- Switch to root by `su`
- Add key to home directory of root like described in documentation with command `pamu2fcfg` in capter Setting up key.

Hope that work.

usrstephan, 2020-05-04:
su user only changes the user until you exit. It does not switch to root. And for this change of the user I need to know the PAM-file.
In my case the normal user is not part of the sudo group - for security reasons as several employees use this account...

nickray, 2020-05-04:
The following works:

```
pamu2fcfg -u root | sudo tee -a /etc/u2f-keys
printf "\nauth required pam_u2f.so debug authfile=/etc/u2f-keys\n" | sudo tee -a /etc/pam.d/su
```

I'd keep that `/etc/pam.d/su` file open in an editor with write access to it :)

Once it works, remove the `debug` argument.

nickray, 2020-05-04:
I still think you should restrict `sudo` rights to this shared user to require the key, instead of using `su` :)

nickray, 2020-05-04:
Looks like there's FIDO2/RK integration, but not released yet.

jolo1581, 2020-05-05:
@nickray fido2 might not Work, but you can configure the pam module to set the Key required.

nickray, 2020-05-05:
I do believe you cna use RK+PIN if you compile / install from head of the repo 👍

usrstephan, 2020-05-06:
thanks nickray - I did not test the whole workaround you described, but /etc/pam.d/su was the key. I added pam_u2f.so and it worked as a charm:-). I did not completely understand the part with the format... As I only copy pamu2fcfg into the directories of my sudo user restrictions should be sufficient.
thanks very much for the fast and helpful comments on this forum!!!
two remarks:
- about su: I have two accounts: one with sudo access and one without; after configuration the sudo access is mainly needed for update checks. As my machines run under the non-privileged account all day, the easiest way for me is to use su WITH the name of the privileged user. So I believe I do not use the root account and do not see security problems even if it may be crufty:-). Please let me know if I have missed any important aspect
- is there any tutorial about RK, so I can take a look

closed 2020-05-06 by usrstephan

nickray, 2020-05-06:
Regarding the format: Since credentials are not transferable between devices, and your key may be lost or break, you should add at least two credentials on two separate keys (main + backup).

I don't have the time or interest to go into my views on sysadmin best practices.

For RK, https://www.w3.org/TR/webauthn/#resident-credential is maybe an entry point. Whereas U2F was a second factor (to password), and the relying party selects potential credential IDs via known username (allowing the key to not actually store the credential, and instead encode it in encrypted from in the credential ID), with FIDO2 resident keys are a) stored on the device, and b) passwords are no longer strictly necessary. Instead, there's a "PIN" (which doesn't have to be numerical, and is only transmitted to the key to unlock it, *not* to the server/relying party) to protect the key, and relying party reads out possible credentials from the key. In this way, it's still two-factor authentication (key is something you have, PIN is something you know).

usrstephan, 2020-05-06:
of course I use 2 sticks!
I will have a look on RK
thanks a lot

jolo1581, 2020-05-09:
@nickray I am also interested on Linux login with solo only.
Which repo I have to install? libpam or libfido2?

nickray, 2020-05-09:
I'm not aware of passwordless login being possible (yet).

jolo1581, 2020-05-09:
I mean key + pin

nickray, 2020-05-10:
Yeah but then the login manager would need to a) know which credential ID to use, or b) give you a selection window to pick a resident key and enter PIN. I don't know of sucg a login manager.

usrstephan, 2020-05-10:
@jolo1581
you find a great tutorial here: https://schulz.dk/2019/08/23/using-solokey-for-linux-login/
for terminal authentication you need to edit /etc/pam.d/sudo and /etc/pam.d/su
if you place auth sufficient pam_u2f.so BEFORE @include common-auth in these files you can login with solokey only. in case your key is not in place you can use keywords... like before
This works great for me.

jolo1581, 2020-05-10:
@usrstephan 
I wrote the documentation for using solo as login token in solo documentation.

But I havn't found any information about login with resident key and pin. I am interested, because Windows Hello works. 
I would like to test it on Linux and then add all information to solo documentation.