solokeys/solo1 #434: Documentation pages not rendering properly

Iolaum, 2020-05-08:
Something looks wrong with the documentation pages. They are not rendering properly on both Firefox and Chrome (latest versions on Fedora MATE 32). I am attaching a full page Firefox screenshot of how the https://docs.solokeys.io/solo/programming/ page looks on my machine. (On Chrome it looks the same as the Firefox screenshot.)

[image]

jolo1581, 2020-05-08:
Thought @nickray repaired it a few weeks ago.

Iolaum, 2020-05-09:
Page renders properly now. :+1:

Iolaum, 2020-05-16:
The issue has re-appeared.

nickray, 2020-05-16:
yeah, sorry :/ i'll fix this for good when i can get to it.