solokeys/solo1 #436: SiFive based RISC-V targets

LeonardKoenig, 2020-05-12:
Hello,

we are a group who is interested in adding another target for solokeys, namely RISC-V boards by SiFive. We are at the start of the project, however we'd wish to eventually upstream the changes. In order to coordinate this, I'd like to ask about your preferred way of adding another target.

The SiFive boards are supported with the Freedom E SDK which comprises the Freedom Metal library (common code) and Board Support Packages which do the actual wiring up of the hardware with Freedom Metal. I'd propose adding something like

    targets/freedom-e/lib/freedom-metal [git submodule]
    targets/freedom-e/lib/bsp/... [imported from the SDK]

As the layout for that platform. However, freedom-metal is a rather complex library which needs autoconf for setup and detection, while the current STM32/CMSIS target simply "build". This would mean we'd have rather different build-systems for different platforms, which may be undesirable. The upside of this would be less maintenance needed for the freedom-e boards, as we "simply" would need to update the git submodules (if everything works fine). The alternative would be to create a "standalone project" using `make standalone` in the SDK and importing these files manually into the source tree. This would mean standalone projects for each specific SiFive board, eg.

    targets/sifive-hifive1-revb/lib # contains standlone freedom-metal files pre-configured for the platform

I'd like to know your opinion on which approach you'd prefer.

nickray, 2020-05-12:
No personal opinion on this. Just a heads up that the upcoming v2 key will run a v2 firmware which is a rewrite/redesign in Rust. Bringing this up as you state you're at the beginning of your port.

LeonardKoenig, 2020-05-17:
@nickray thanks for the quick response, sorry I didn't have the mind to reply earlier. I couldn't find any other information on this except for the issues here, but I guess the current implementation is in-house only. I'm a bit curious whether you have already evaluated building Solokey V2 on the infrastructure google created with TockOS+OpenSK?  This is a rust-based opensource implementation of FIDO2 (although not yet certified). As it's uses the "fullfledged" TockOS instead of a simple STM HAL, it might be easier to port.

For us though, as we are planning to do this as part of a university project, we will probably now evaluate a Solokey C port or helping use OpenSK on the HiFive (an initial TockOS port is already there).

nickray, 2020-05-19:
Hey @LeonardKoenig, yes we evaluated TockOS and don't think it's suitable. We are also in contact with the team at Google, and while it makes sense for them, it doesn't for us. 

Feel free to chat me up on Keybase (@nickray) if you'd like to some opinions :)