solokeys/solo1 #437: Import PGP sub-keys

MaxHillebrand, 2020-05-15:
I would like to import my already existing PGP sub-keys onto the solo key. 

However, after I plug in my solo key, when I run `gpg --card-edit`, it returns `selecting openpgp failed: no such device. OpenPGP card not available: no such device`.

I have a debian 10 virtual machine, with these additional programs installed `wget gnupg2 gnupg-agent dirmngr cryptsetup scdaemon pcscd secure-delete hopenpgp-tools yubikey-personalization`.

What software setup do I need in order to import my PGP keys?

nickray, 2020-05-16:
Solo doesn't currently have any PGP support.

MaxHillebrand, 2020-05-17:
Oh, then I'm sorry I thought it had it... 
Thanks for clearing it up.

closed 2020-05-17 by MaxHillebrand

MaxHillebrand, 2020-05-17:
but may I say that this would be a nice feature to add, if possible.

merlokk, 2020-05-17:
https://github.com/solokeys/openpgp

slow coming feature...