solokeys/solo1 #441: [U2F] U2F Error code 6985 with pamu2fcfg

alinelena, 2020-05-23:
burned the current git firmware on a nucleo32... 
I could test the u2f websites and fido2 all seems ok... 

now i was trying to setup sudo as indicated here

https://schulz.dk/2019/08/23/using-solokey-for-linux-login/

when i try to enroll things appear to work correctly but in logs for the solo i see

```
[U2F] U2F_REGISTER
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[U2F] U2F_REGISTER
[U2F] sha256: cc 10 79 c9 3b c5 10 4f e0 e5 ab 4f a5 12 60 85 93 aa c4 fd d4 51 1c 71 a3 3a 68 16 cb 91 96 84 
[U2F] u2f resp: 05 04 6e .............
```

of course sudo does not work after

```
[EXT] Ignoring U2F check request
2c 60 7f c1 a1 6f de c4 ab 8f 53 9f b0 c0 95 03 11 d4 f8 7f 00 d3 4a f7 9e 36 9e 92 eb 14 d7 06 b7 cd 81 c0 22 f3 59 19 2f 8a 36 88 8c 7e 51 cc 
[U2F] checked CRED SIZE 48. (FIDO2: 70)
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[EXT] Ignoring U2F auth request
2c 60 7f c1 a1 6f de c4 ab 8f 53 9f b0 c0 95 03 11 d4 f8 7f 00 d3 4a f7 9e 36 9e 92 eb 14 d7 06 b7 cd 81 c0 22 f3 59 19 2f 8a 36 88 8c 7e 51 cc 
[U2F] checked CRED SIZE 48. (FIDO2: 70)
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[EXT] Ignoring U2F auth request

```
this is on the master head cc4cd340da86ea7340792f785538b4896e460864

pam_u2f version is 1.0.8

any idea what can go wrong?

jolo1581, 2020-05-23:
You might take a look to [https://docs.solokeys.io/solo/application-ideas/](https://docs.solokeys.io/solo/application-ideas/)

alinelena, 2020-05-24:
@jolo1581 thank you... success or sort of..
this is on opensuse tumbleweed

played to get the sudo working and indeed works... had to do this
```
circassia:/home/drFaustroll # cat /etc/pam.d/sudo 
#%PAM-1.0

auth  required       pam_u2f.so authfile=/root/.config/solokeys/sudo  
auth include common-auth 
account     include     common-account

```

a little bit complex but works..

for session login was simpler... just added
```
auth  sufficient       pam_u2f.so
```

before
```
auth include common-auth 
```
now the strange thing is where it read the u2f keys from... root for sudo home from session... which makes sense I think.

now despite all working i still see the error
```
[U2F] checked CRED SIZE 48. (FIDO2: 70)
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[EXT] Ignoring U2F auth request
```