solokeys/solo1 #442: Any news about SoloKeys v2?

screendriver, 2020-05-24:
First it would be great to have [GitHub discussions](https://help.github.com/en/github/building-a-strong-community/about-team-discussions) enabled because sometimes users don't report issues. Instead they want to ask something or start discussion. Like me now 🙂 

I just wanted to know how far the development of SoloKeys v2 is right now? I'm asking because maybe I want to give away some keys as a birthday present and maybe introduce SoloKeys within my company. And instead of giving away a (soon?) "old" version of a key that should last a very long time I'm asking myself if it's worth it to wait a little bit longer until v2 came out?

nickray, 2020-05-24:
Hey @screendriver 

- I'm trying to get Discussions actually (to move this kind of thing off of Issues, which should be for bug reports), currently the application process is fully booked. This is tied to a repository though, I wonder if we should have a dummy "Community" repository and tie it to that? Given we have a bunch of repositories (and will have more), and I don't think it's reasonable to manage N such Discussions. Thanks for the suggestion!

- Generally, v2 is "done when it's done", we want to get this right vs. rushing it out (of course, we're excited ourselves to get it out). Regarding "old" keys, they'll definitely be continue to be supported with bug fixes, we're seeing new patches like @rgerganov's coming in, and hopefully PGP will complete sometime too. I'd say for the birthday present, giving a Solo A (no Tap) is the safest option, as it has full support for all current and future CTAP extensions, and is FIDO certified. Solo Tap if they have a phone with good NFC support.

Regarding state of development, most things are mostly working, but there's a bunch of refactoring, testing, functionality to add before we want to make things public. So something like "80% done", with the remaining 20% being another 80%, if you know that reference :)

For your business use case, feel free to send us an email at hello@solokeys.com, or chat me up as @nickray on Keybase directly.

erazemkokot, 2020-05-24:
> I'd say for the birthday present, giving a Solo A (no Tap) is the safest option, as it has full support for all current and future CTAP extensions, and is FIDO certified. Solo Tap if they have a phone with good NFC support.

Regarding the CTAP (and any other) extensions/features (like OpenPGP/TOTP), where on the "safe spectrum" would you put the SOMU?

nickray, 2020-05-24:
Somu has the same firmware as the regular sized keys, all currently available keys are equally future proof in that sense.

screendriver, 2020-05-24:
Thank you for the very detailed answer :+1: 

> This is tied to a repository though

Oh, this is new for me. I thought this is for the group and not just for a single repository. Hm then maybe a dedicated community repository or a monorepo? :thinking: 

So I will close this for now because everything was answered :sunglasses:

nickray, 2020-05-24:
I pinned it as I think this will come up more and more the closer we get to potential release :) So if you don't mind, I'll reopen. Thanks for asking!

screendriver, 2020-05-24:
Oki doki :ok_hand:

coelner, 2020-09-28:
https://solokeys.com/blogs/news/long-awaited-update

michaelblyons, 2020-12-02:
I didn't miss the Kickstarter, did I? Did it happen already? 😞

patschi, 2020-12-02:
> Did it happen already? 😞

Nope. Quoting Solokeys from [Twitter here](https://twitter.com/SoloKeysSec/status/1326672780612481024):
```
Yes...we got overly optimistic.  We make great hardware, but
horrible predictions.  We will have news to share in December.
```

M0NsTeRRR, 2021-01-03:
https://solokeys.com/blogs/news/a-new-year-a-new-launch-solo-v2

M0NsTeRRR, 2021-01-26:
Kickstarter => https://www.kickstarter.com/projects/conorpatrick/solo-v2-safety-net-against-phishing

msiqx9, 2021-02-12:
> Kickstarter => https://www.kickstarter.com/projects/conorpatrick/solo-v2-safety-net-against-phishing

I went to this link with the intention of supporting Solo v2 but I saw an offensive "# BLACK LIVES MATTER" banner at the top of the page.  Sadly, I had to move on.

erazemk, 2021-02-12:
> I went to this link with the intention of supporting Solo v2 but I saw an offensive "# BLACK LIVES MATTER" banner at the top of the page. Sadly, I had to move on.

You know that's from Kickstarter not Solokeys, right?

michaelblyons, 2021-03-19:
Issue can be closed now? Or link IGG?