solokeys/solo1 #449: Unable to use Solo Key with Duo (Chrome 83, macOS 10.15)

GalaxyAllie, 2020-06-09:
I get various errors when trying to use a Solo Key with Duo. Sometimes, it doesn't see it at all, sometime it fails to authenticate and I get locked out. Right now, I can't even re-enroll after unenrolling to try and fix the problem, I only get 'WebAuthn enrollment failed.'

Using the [Duo-designed webauthn.io](https://webauthn.io) I get the following error:

`The operation either timed out or was not allowed. See: https://www.w3.org/TR/webauthn-2/#sctn-privacy-considerations-client.`

The key has firmware 4.0.0 and I have completely reset it a couple times trying to get it to work...

My1, 2020-06-10:
granted I am on linux, but it all works so I have a few questions:

1) is a PIN set on the Solo
2) secure or hacker (on a hacker a logging build may help)
3) did you set any specific parameters on the site?
4) you might also try around on this little page I run and play with the parameters
https://my1.dev/wa/my1.php
because there one can literally see the entire build of the parameters (and can hack in even more to pull the outputs nicer and whatever)

GalaxyAllie, 2020-06-10:
Hi, I've tried on my Chromebook as well and all the same problems. It's a secure, and I've tried both with and without parameters and with and without a PIN set after resetting the key in various combinations. I've tried your page and am not quite sure what I'm looking at (sorry, I'm pretty new to WebAuthn) - but quite a few times it doesn't even respond to my taps.

I'm tempted to close this issue because if the product was this broken, there would be mass complaints - I think I've just managed to break my key somehow :( - with more experimentation, it isn't Duo specific. It now works on very few sites (it does work on the [Yubico test](https://demo.yubico.com/webauthn-technical/registration)). It was fine before the 4.0.0 firmware (I think, maybe earlier - it's a backup, I usually use my YubiKey), but this may be coincidental.

I think when I have some spare time I'll have to figure out how to reflash the key - if I even can (I'm not sure what protections against reflashing are in the secure version, but I suspect I can't... or will at least have to work at it...)

conorpp, 2020-06-11:
Strange, I would expect Duo's WebAuthn site to work fine.  

How are you resetting the key?  
Is this over USB?  
Is there another authenticator built into your laptop (e.g. TPM)?  
Does the LED on the device change color when a request is sent?
What browser?

aseigler, 2020-06-11:
FWIW, I just tried registering Solo 4.0.0 using USBA on https://webauthn.io, macOS 10.15.4 on Chrome 83.0.4103.97 with attestation type "none" and authenticator type "unspecified" and it worked perfectly.

GalaxyAllie, 2020-06-11:
Thanks @aseigler 

@conorpp - with the reset command in solo-python (`solo key reset`), yes it is over USB, yes there is Touch ID on the Mac - the Chromebook has built-in U2F support but it's behind a feature flag I have disabled, yes it turns orange, and Chrome 84 beta on both :)

conorpp, 2020-06-11:
Ok when it turns orange, and you press the button-- does it quickly blink teal and back to green?  Is there a possibility you're sometimes removing power when pressing the button?

GalaxyAllie, 2020-06-11:
@conorpp aha! I suspect you're indeed correct that it is a power issue. I just tried on my ThinkPad and... it worked. I had to push the button a few times to login though (but not to register). The PIN prompt came up instantly, it was after the PIN prompt it became tricky.

However, I tried again with the built-in NFC - which *has* worked for me in the past on the ThinkPad. After entering my PIN, when I tapped again, Windows started rapidly playing NFC boop tones... like I kept removing it and putting it back multiple times per second. Needless to say, login did not succeed and I saw the same behavior I saw over USB on my Mac and on my PixelBook.

I will go ahead and close this as I think I can put it down to defective hardware...

closed 2020-06-11 by GalaxyAllie

conorpp, 2020-06-11:
How long are you leaving it on the NFC?  Depending on how many credentials are registered, if a PIN is set, it can take multiple seconds with Windows

GalaxyAllie, 2020-06-11:
@conorpp just a few seconds but it's constantly disconnecting and reconnecting so it's not actually sustaining communications. And this is on a freshly reset key...

My1, 2020-06-11:
okay interesting, although the Solo's NFC is a bit inconsistent across devices. my Solo C Tap has no problem on the NFC reader I use sometimes (reinerSCT cyberjack basis)