solokeys/solo1 #451: Error when running `solo key verify` (Somu, Windows 10, Python 3.7.7)

dannya, 2020-06-17:
Received a Solokey Somu today, encountered an error on running `solo key verify`:

```
C:\>solo key verify
Please press the button on your Solo key
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2032.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.2032.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\Scripts\solo.exe\__main__.py", line 9, in <module>
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\solo\cli\key.py", line 323, in verify
    cert = solo.client.find(serial, udp=udp).make_credential(pin=pin)
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\solo\client.py", line 46, in find
    p.find_device(dev=raw_device, solo_serial=solo_serial)
  File "C:\Users\d\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\solo\client.py", line 104, in find_device
    raise solo.exceptions.NonUniqueDeviceError
solo.exceptions.NonUniqueDeviceError
```

Note that whilst the log shows `"Please press the button on your Solo key"`, the errors happen instantaneously, there was therefore no time to press the button.

------------------------

Output of `solo ls`:

```
C:\>solo ls
:: Solos
208839AE384B: Solo 3.0.0
```

------------------------

My environment:

```
Windows 10 Insider (Build 19645)
Python 3.7.7
```

_Note that I also have a Yubikey 5 Nano plugged in concurrently._

nickray, 2020-06-18:
Dupe of

nickray, 2020-06-18:
Dupe of https://github.com/solokeys/solo-python/issues/55

closed 2020-06-18 by nickray

dannya, 2020-06-18:
@nickray IMO not an exact dupe, since in the linked issue at least they are seeing a handled, user-friendly error message when running their command.

sevanteri, 2020-07-08:
I get this too when I also have my Yubikey inserted. After removing the Yubikey I can run update, verify and others.

My setup: Ubuntu 19.10, SoloKeys Solo Tap 4.0.0, Python 3.7.5.
And Yubikey 4 Nano.

So yeah, I'd say this should be re-opened.