solokeys/solo1 #453: USB HID over UDP (No Hardware Needed) not working

up201804351, 2020-06-22:
I am trying to implement a virtual authenticator FIDO2, but the browser does not identify the virtual device created in the module that sends USB HID over UDP (No Hardware Needed).

What is needed for this to work?

kyleharte, 2021-01-26:
It defeats the purpose of a hardware security key. A security key is something you should always have physical access to.

nickray, 2021-01-26:
I think that won't work - the browser expects to talk to USB, while your virtual authenticator would listen on UDP.

enrikb, 2021-01-27:
The 'hidg' mode might work using the PC version of the solo software on Linux, emulating a virtual USB gadget. I had it working in the past, but can't remember if I tested it with a browser. I will give it a try again if I find the time.

enrikb, 2021-01-27:
Yes, the PC version of the solo firmware also works with a browser if used in 'hidg' mode on Linux.
If you manage to compile, maybe sign and load/configure the `dummy_hcd` module found in `tools/gadgetfs`, you can start the virtual solo as `main -b hidg`. All this usually needs to done as `root`. Then:

1. Verify to find the virtual device in `lsusb` output
2. Verify that `/dev/hidrawN` (N == 1 in my case) exists and is accessible
3. Verify that `fido2-token -L` shows the device

If this all works well, it should also work in the browser e.g. using [WebAuthn.io](https://WebAuthn.io/).