solokeys/solo1 #455: solo.exceptions.NoSoloFoundError: no Solo found

tarun14110, 2020-06-25:
The solokey was working fine. I was playing to build and update the firmware using the instructions from the documentation. The solokey stopped working after that. No LED blinking, no output in `dmesg`.  Is there a way to reset everything?

solo program aux enter-bootloader
> Traceback (most recent call last):
>   File "/home/tarun/.local/bin/solo", line 11, in <module>
>     sys.exit(solo_cli())
>   File "/home/tarun/.local/lib/python3.6/site-packages/click/core.py", line 829, in __call__
>     return self.main(*args, **kwargs)
>   File "/home/tarun/.local/lib/python3.6/site-packages/click/core.py", line 782, in main
>     rv = self.invoke(ctx)
>   File "/home/tarun/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/home/tarun/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/home/tarun/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/home/tarun/.local/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/home/tarun/.local/lib/python3.6/site-packages/click/core.py", line 610, in invoke
>     return callback(*args, **kwargs)
>   File "/home/tarun/.local/lib/python3.6/site-packages/solo/cli/program.py", line 219, in enter_bootloader
>     return _enter_bootloader(serial)
>   File "/home/tarun/.local/lib/python3.6/site-packages/solo/cli/program.py", line 200, in _enter_bootloader
>     p = solo.client.find(serial)
>   File "/home/tarun/.local/lib/python3.6/site-packages/solo/client.py", line 52, in find
>     raise solo.exceptions.NoSoloFoundError("no Solo found")
> solo.exceptions.NoSoloFoundError: no Solo found

conorpp, 2020-06-25:
It's probably currently in DFU mode.  You can program a new bundle firmware, and then run `solo program aux leave-bootloader`.  If you never programmed anything, you can just run that leave command.  Just be careful not to leave the DFU mode when nothing is programmed, or then the key will be bricked.

tarun14110, 2020-06-25:
@conorpp Thanks for the quick reply. On trying `solo program dfu <all.hex>` , I get  `raise Exception("no DFU found") Exception: no DFU found`

conorpp, 2020-06-25:
You need to have libusb installed.  Do you know if you have this?

tarun14110, 2020-06-25:
Yes! I have installed libusb.

conorpp, 2020-06-25:
Can you share some more info about your OS, and try seeing any system logs for when you plug in the device (e.g. `dmesg` on macOS or Linux).

tarun14110, 2020-06-25:
I don't see anything in 'dmesg' on plugging and unplugging. Earlier I could see the details of the device. Now I see nothing. Does that mean I break the solokey?

conorpp, 2020-06-25:
Yeah, I'm not sure, other than checking that the power is making contact in the USB port.  If it's an earlier USB-C model, then it could be some of the pins on the USB-C connector cracked, and could be reflowed with a soldering iron.

tarun14110, 2020-06-25:
It's a USB-A. I bought it recently a couple of months ago. I am ordering another solo key so that I can continue on my work. But, I would like to debug or at least find the cause of the issue with this solo key. I successfully loaded the program a few times using `solo program bootloader <>` and then suddenly it stopped working. Can you point me to some resources that can help me to debug?

Pavithra-ar759, 2025-03-14:
try activating your virtual environment and try giving the command 

source venv/bin/activate