solokeys/solo1 #457: Please support multiple identical keys

BillDietrich, 2020-06-30:
I want to buy or make three to five identical keys. I've been told this is explicitly against the FIDO/U2F standard. But I want it. I want to put one or more of those keys into a safe-deposit box, yet still be able to register/unregister it (all keys in one operation) to/from accounts without having to retrieve it out of safe-deposit each time. I want to put one key into my PC, authorize it to an account, and have all 3 to 5 identical keys work on that account.

I sent email to the SoloKeys contact address, and the response said a firmware modification would be required to do this. It also said "The firmware for Solo Hacker is unlocked, so you would be able to add/modify firmware as appropriate."

Has anyone done such a modification, or have it in progress ?  Please consider adding it as a standard feature, not even requiring Solo Hacker.  Thanks.

JohnVillalovos, 2020-07-09:
That won't work. The keys have a counter. The server keeps track of the last counter used. If the counter ever goes backwards most likely the server will invalidate that key and all of its clones.

It is discussed in this video: https://youtu.be/DWrLBwi7ZBA?t=1048

BillDietrich, 2020-07-09:
Thanks for the info.  Yes, someone else said same in another forum after I posted similar request.  Bummer.

Registering multiple different keys does not satisfy my use-case.  I want to keep one or more backup keys in off-site storage (safe-deposit, leave with a relative, etc).  Then just use one key daily.  Any time I create a new account or an account adds support of 2FA, I want to apply/register that one key to the account and now have ALL of my keys work on that account.  I don't want to have to retrieve all the other keys out of storage and apply all 3+ keys to the account.

MaPePeR, 2020-07-09:
Maybe this article helps: https://dmitryfrank.com/articles/backup_u2f_token

BillDietrich, 2020-07-09:
Okay, thanks very much for that.

If I'm reading that strategy correctly, as soon as you use a backup token, the counter gets high and your primary token stops working.  So you would have to program say 3 tokens to be used in a specific order, with increasing starting values for the counter: use A until you lose it or it breaks, then you switch to B and use it until you lose it or it breaks, then you switch to C.

DanielG, 2020-11-22:
Is the counter staying confidential actually critical to the security of the system? What if one were to simply store/sync the counter on the PCs the keys will be used from.

You'd probably need a small daemon to talk to the key in order for the PC to be informed when the counter changes and restore it when the key is plugged in though.

nickray, 2020-12-08:
To clarify the counter issue: In U2F, it is required, in FIDO2 (which supersedes U2F and browsers actually implement), the standard allows always using counter = 0 (https://github.com/dicekeys/seeding-webauthn#setting-the-signature-counter). Of course, it is up to the relying party to enforce whatever behaviour they wish, although in practice it seems as though nobody checks the counter.

Decstasy, 2021-01-12:
Please correct me if I'm wrong...
I can program two identical keys and I could flash the second key with a firmware, which has a static counter increase. The --lock would protect my private key but not the bootloader nor just accepting signed updates.
Is there a way to lock the hacker edition completely (including the bootloader) after I created the keys?
Or the possibility to trigger an erase when someone tries to flash the device again?

Ciantic, 2022-08-13:
Counter only needs to be bigger than last result. Simple hack is to issue current UNIX timestamp as counter on each cloned key.

This way you can use multiple keys as each gives a latest timestamp that is greater than last one.

PinkysBrain, 2022-12-10:
Given that with passkeys, Google/Apple and soon Microsoft are going to be merrily cloning, is it perhaps time to revisit this for Solo? Maybe a flag set during initialization to allow physical cloning with a *gasp* password passed by the cloning device? :)

It doesn't make much sense to use the standard as an argument for "physical" keys when the Fido alliance is allowing the big vendors to just forget about it for their secure enclave implementations.

szszszsz, 2022-12-10:
@PinkysBrain What do you mean by cloning? Can you point to an article or place in the specification which claims that?