solokeys/solo1 #460: Can we make a Solo SoftKey?

hey24sheep, 2020-07-04:
Hi, I am kinda new to this ctap and stuff. I am researching for a few days now. I couldn't find a software authenticator. Close to what I am asking for is this library by [Github SoftKey](https://github.com/github/SoftU2F) which is only for MacOS. I currently have a U2F key but it is not supported by Microsoft Azure Login. 

Also, I want flexibility of using nearest device to authenticate instead of pushing a button on my key.
I have a few doubts and ideas. I hope someone can clear my doubts and correct me. Thank you.

Idea
- Use my android fingerprint to generate a pin to authenticate browser webauthn.
- OR use a desktop program which could auth the request using key generator or biometric

Doubts
- Is it secure enough to use software authenticator (other than getting hacked, I mean)? Why no one is using them like password managers?
- Can we use this firmware on Arduino Uno? as it is similar to Nucleo. As I have a spare arduino uno :)

Thank you for helping me here.

merlokk, 2020-07-04:
webauthn on android works with fingerprint. you can test it with test page and google crome in android phone.

software authenticator cant be secure enough... because there are many ways to grab ram from another thread/process/dll injection/ ring-0 access / etc

Arduino UNO have another CPU  https://en.wikipedia.org/wiki/Arduino_Uno
and it 8-bit atmel. stm32 can be used in arduino editor but its another thing...
it needs to rewrite some code to make it work on 8-bit cpu and it will be very slow.. 

good idea to use phone and its security layer as FIDO2 authenticator..... but i dont see here a place for code from this repository)
but) it will be cool)

merlokk, 2020-07-04:
BTW. 
there is an USBIP technology for linux. and it works pretty good. So you can plug key on one pc and use it from another. Need to think about button, but Solo Hacker can be recompiled with disable the button)

hey24sheep, 2020-07-04:
@merlokk I see. My main thing is, okay that I have a key but I want it as a backup key. I do not want to move and go press a button on my pc or usb hub. I want to emulate a key based on biometric or something else. From my desktop or mobile. 

Security issue that you have stated could occur with software when they are cached or saved. But what about if it mimics the hid as if it is a virtual hid fido device?

I see that now on latest updates android shows a fido security key bluetooth button in notification panel. Which is used by Google at least for now. So from my pc via bluetooth gmail sents notification to my phone for authentication. Which could be neat if it could be used by other websites like website -> key popup -> solo ble -> android. Here solo ble interprets android response and sends it back to os as virtual hid result. I mean I'm not a system level programmer so I do not know if this is just plain stupid or not. But something based on GitHubs soft key plus solo.

coelner, 2020-07-21:
1. You should have a real backup key, e. g. a second one
2. this softkey uses somehow the internal macOS security features, which relies usually on a crypto hardware. That seems legit in the apple environment. But do not assume this is only software and runs on a stupid cpu (like STM32 or ATMega). But microsoft azure login needs real hardware with fido compliance
3. Are you sure that this happens all over bluetooth? I don't think that a FIDO device is connected simultaneous to two devices.  My understanding is that you can pair a BLE FIDO device with a google account and use it with the built-in features of android or an additional app under iOS. Then you can use a chrome based browser to get over your google account a relay to the google account connected app which get informed with a google service based push notification about a fido request. ( I assume this a workaround for computers which do not have a possibility for nfc/usb/bluetooth , https://www.ftsafe.com/download/webdownload/ePass%20FIDO-Multi-Interface%20Manual.pdf)
4. the modern google Pixel phones do include such a fido2 device, most devices do have a hardware based crypto device (like tpm for pc) which could be used, if the vendors implements the api within their firmware. It will become more popular.
5. If you want to use just simple software, then use a 2FA application
6. https://techcommunity.microsoft.com/t5/identity-standards-blog/all-about-fido2-ctap2-and-webauthn/ba-p/288910
7. https://github.com/google/OpenSK

-----------------
However, I'm not sure what you are trying to do. 
 - Either you have one 2FA hardware device, which you takes everywhere you go OR you have for each device a 2FA device
 - You need a backup 2FA hardware token OR at least a secure offline storage for the backup codes
 - You can disable the button, but maybe the logins do not work then (yes, you can patch this, but why then a second factor authentication? )

hey24sheep, 2020-07-21:
@coelner I think you are understanding it wrong.

Here check this [google](https://support.google.com/accounts/answer/9289445?p=phone-security-key&visit_id=636910140416317993-4117575157&rd=1) link. Now do you understand what I am saying? I have a FIDO Key [hyper fido mini](https://hypersecu.com/tmp/products/hyperfido). 

Let me clear some things here.
- I do not own 2 keys, they cost a lot in INR. I own one
- Disabling the button is far less of an issue. It's multiple hardware and which is not guaranteed to work on every website
- Even after spending money, Azure doesn't support this key instead they support other ones like yubikey which are almost 6-7k INR or 90-100$ plus.

Here are some screenshot of this Bluetooth Service on my Phone (Huawei Honor Play, Android 9)

[image]
[image]
[image]

Now in respect to this + Github's MacOS SoftKey. Why can't we use Windows or Android's built-in key generation for CTAP2? I mean this. I am work from home, Why should I reach out to my PC to click a button when I can do it from my PC or Phone. I know some might say it's useless or not secure enough or something else. But it's perspective. I mean why not? I carry my key if I go to office. I should have soft + hard key option. Have you heard of Card-Less ATM? People might say why? but, Why not?

And I know this might not work on Azure or other platforms like Microsoft. But, I feel it is a starting point in unifying stuff. Why should I buy another hardware which I definitely have to replace in few years to update security and carry multiple things with me.

coelner, 2020-07-21:
Yeah, you know, the features in Android / windows are based on a crypto device. You need to interact with the hardware device (separate dongle, smartphone, integrated whatshowever in the x86) to legitimate a specific transaction. Otherwise you could use a simple arduino uno which parse a static string. 

As I understand you are lazy to reach this button. Use an esp8266 and build a button remote. (The hyper fido mini device can be opened, it contains a infinion chip, but the button could be triggered by an esp8266 gpio. E.g. use two of the esp8226 and the espnow implementation. maybe similiar to this https://revspace.nl/EspNowSkip )
Why don't use your smartphone as additional u2f/fido device? Maybe it supports even fido2?

The hyper fido mini is only FIDO (aka U2F aka CTAP***1***), not FIDO2 (aka CTAP***2***). the same is for MacOS SoftKey, it is only a U2F.
As I mentioned there is a implementation for an nRF chipset. Otherwise it should be portable to arm trustzone, amd psp, intel sgx, apple t2. Why didn't you use this firmware and run it in software like mentioned here: https://github.com/solokeys/solo/blob/master/README.md#developing-solo-no-hardware-needed and get for your solution a certificate (https://fidoalliance.org/certification/authenticator-certification-levels/ and this https://fidoalliance.org/certification/authenticator-certification-levels/ )
I don't know the amount of money you need to get the lowest certification.

hey24sheep, 2020-07-21:
> As I understand you are lazy to reach this button

No, its not just that. It's inconvenience of plugging it into the desktop and mainly the cost of FIDO2 devices in India. 

> Why don't use your smartphone as additional u2f/fido device? Maybe it supports even fido2?

Cuz, I can't no one supports it except google. I can't seem to find any API on Android for that. 

> Why didn't you use this firmware and run it in software like mentioned here

I will try that, I haven't got time to fiddle with this. I thought someone else might have already done it so I opened an issue here first

Weird thing is, If you check [this](https://fidoalliance.org/fido2/) flow diagram. It states internal authenticator, but no mention in docs.

coelner, 2020-07-22:
I think you need dig in the google service api, it is not a aosp feature.

Yes it says internal authenticator...like a crypto chip which is certified by the fido alliance.