solokeys/solo1 #462: want to use Python to write a firmware for the fido standard hardware token.

IR-Consensus, 2020-07-13:
Hello.

I want to use Python to write a firmware for the fido standard hardware token.

I read the resources in this field,
unfortunately, I could not find a good source.
I want to use the experiences and guidance of friends in this field.
I do not want to do anything wrong.

Like Solokey hardware(use of the components like solo key) 
I want to use supported microcontrollers for this purpose.
( I mean to Use of parts solo key )?

I have several questions?

Is this logical?
Can I use the Python programming language do the Micro Python feature or any library used for programmer hardware?
Does python language able to write firmware with these goals?

I want to use fingerprint chip and CryptoChips and do both with python language.  

Can I get help from firmware solo key for better do? (I don't know this is right?).

nickray, 2020-07-16:
This is not a good idea in my opinion. What might be interesting is embedding a version of MicroPython to allow extending functionality, but this is not on the roadmap at all currently.

IR-Consensus, 2020-07-16:
Thank you for your answer.
Please  explain my mistake.
I want to fix the flaws, Have good growth in learning.