solokeys/solo1 #464: solo key keyboard error

imp1sh, 2020-07-23:
Hi,
I just tried the new keyboard feature but it fails:
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal
~/.local/bin/solo version
0.0.26
uname -a
Linux xps13 5.7.9-050709-generic #202007160732 SMP Thu Jul 16 07:42:55 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
python3 --version
Python 3.8.2
~/.local/bin/solo key keyboard 'test'
Traceback (most recent call last):
  File "/home/jochen/.local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/home/jochen/.local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/jochen/.local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/jochen/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/jochen/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/jochen/.local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/jochen/.local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/jochen/.local/lib/python3.8/site-packages/solo/cli/key.py", line 492, in keyboard
    dev.program_kbd(buf)
  File "/home/jochen/.local/lib/python3.8/site-packages/solo/client.py", line 326, in program_kbd
    return self.ctap2.send_cbor(0x51, cmd)
  File "/home/jochen/.local/lib/python3.8/site-packages/fido2/ctap2.py", line 645, in send_cbor
    raise CtapError(status)
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND
```

Pull request:
https://github.com/solokeys/solo/pull/446 for the keyboard feature

burningalchemist, 2020-08-06:
@imp1sh afaik, it won't work against 4.0.0. Since the interface was implemented in the firmware, you first need to burn the firmware to the chip to make the command work, which is only available for hacker version at the moment.

imp1sh, 2020-12-25:
Is there an estimate when a new firmware version will be released? This feature had been finished some while ago, yet it's not available to the users due to the lack of a new firmware release.

nickray, 2020-12-29:
There are right now no plans for a new firmware version, and I think if/when 446 is merged, it would be an experimental feature (i.e., need to build yourself and flash to a hacker key).

bugzy, 2021-03-26:
Just came in here to inquire about the status of this feature. It looks like the latest python client (v 0.0.27) has the "solo key keyboard" command included, but the latest firmware (v 4.1.2) still does not include the command. Was it just a mistake that #446 did not make it into the latest firmware?