solokeys/solo1 #465: Unable to Update on Ubuntu 20.04

charles-d-burton, 2020-07-31:
Currently it looks like the python library is unable to recognize the solo key for updates in ubuntu 20.04, I've tested it on multiple machines with the result being the same that the device is not found while in bootloader mode.  This is on a system with the latest release of the solo python library as well.

I've followed the instructions found here:
https://docs.solokeys.io/udev/

When I plug in the key in bootloader mode I get this from dmesg:
```bash
[  853.743595] usb 1-3.1: USB disconnect, device number 10
[  858.588314] usb 1-3.1: new full-speed USB device number 11 using xhci_hcd
[  858.913760] usb 1-3.1: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 2.00
[  858.913764] usb 1-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  858.913766] usb 1-3.1: Product: Solo
[  858.913767] usb 1-3.1: Manufacturer: Solo Keys
[  858.913768] usb 1-3.1: SerialNumber: <redacted>
[  858.945011] hid-generic 0003:0483:A2CA.000E: hiddev0,hidraw0: USB HID v1.11 Device [Solo Keys Solo] on usb-0000:03:00.0-3.1/input0
➜  udev git:(master) 
```

But when I run the update I get:
```bash
➜  udev git:(master) solo key update 

No Solo key found!

If you are on Linux, are your udev rules up to date?
Try adding a rule line such as the following:
ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess"
For more, see https://docs.solokeys.io/solo/udev/
```

What's interesting is that when it's not in bootloader mode the python library picks it up and runs fine, but fails because the device is not in bootloader mode.  Meaning the key still works fine on Linux I just can't update/patch it.

jolo1581, 2020-08-08:
Hi

when you put solo into bootloader mode.
Can you Run `solo ls`?

What is the output of that command?

charles-d-burton, 2020-08-10:
When in bootloader mode:
```bash
➜  ~ solo ls
:: Solos
```

When not in bootloader mode:
```bash
➜  ~ solo ls
:: Solos
206C36605548: SoloKeys Solo 4.0.0
```

jolo1581, 2020-08-12:
I tested it in my virtual maschine (Host System ist Manjaro 20). I have no problem to recognize my solo in bootloader mode.
I have not installed the udev rules, because solo should work without them. The actual version of systemd  should regocnize solo without adding additional rules.

Maybe thats your fault. Try to remove the udev rules you installed und check again.

charles-d-burton, 2020-08-14:
The system recognized the solo key fine, it's the updater utility that seems to not.  Even so I removed the udev rules and tried again, still no luck from the updater utility.

charles-d-burton, 2020-08-14:
Interestingly what dmesg reports is pretty different when in bootloader vs normal.  When I plug it in with bootloader mode I get
```bash
[79789.188870] usb 3-1.2.1: new full-speed USB device number 8 using xhci_hcd
[79789.314663] usb 3-1.2.1: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 2.00
[79789.314668] usb 3-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[79789.314672] usb 3-1.2.1: Product: Solo
[79789.314675] usb 3-1.2.1: Manufacturer: Solo Keys
[79789.314678] usb 3-1.2.1: SerialNumber: 0123456789ABCDEF
[79789.323624] hid-generic 0003:0483:A2CA.002B: hiddev4,hidraw9: USB HID v1.11 Device [Solo Keys Solo] on usb-0000:3e:00.0-1.2.1/input0
[79876.235973] usb 3-1.2.1: USB disconnect, device number 8
```

Here is when it's plugged in normal
```bash
[79880.328389] usb 3-1.2.1: new full-speed USB device number 9 using xhci_hcd
[79880.458050] usb 3-1.2.1: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
[79880.458056] usb 3-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[79880.458060] usb 3-1.2.1: Product: Solo 4.0.0
[79880.458062] usb 3-1.2.1: Manufacturer: SoloKeys
[79880.458065] usb 3-1.2.1: SerialNumber: 206C36605548
[79880.467495] hid-generic 0003:0483:A2CA.002C: hiddev4,hidraw9: USB HID v1.11 Device [SoloKeys Solo 4.0.0] on usb-0000:3e:00.0-1.2.1/input0
[79880.517563] hid-generic 0003:0483:A2CA.002D: hiddev4,hidraw9: USB HID v1.11 Device [SoloKeys Solo 4.0.0] on usb-0000:3e:00.0-1.2.1/input0
```

jolo1581, 2020-08-14:
Well I tried on Ubuntu Live System.

[image]

I have no problems to regocnize solo in bootloader mode. With solo ls i see the bootloader version.

Did you lock your bootloader permanently? So update is not available any more?

charles-d-burton, 2020-08-14:
No, I'm saying it recognized the Solo key just fine in dmesg in bootloader mode.  That's never been the issue, the problem is that the update utilities do not work.

jolo1581, 2020-08-14:
When solo bootloader is not displayed in solo ls, it must be a problem with your solo.

Take a look to my dmesg output. The output says as product: solo bootloader 2.3.0. In your output it says only solo

charles-d-burton, 2020-08-14:
I've tried multiple keys, 3 of them.  None of them are recognized by the updater utility when in bootloader mode.  The OS detects them just fine, they work fine for normal use, I just can't update them.

jolo1581, 2020-08-14:
This is, because your solo bootloader hasn't been recognized.

When I use the command solo ls I see also serialnumber, name solo bootloader and softwareversion of bootloader.

I can update my solo in bootloader mode with solo key update

I tested it with Ubuntu 20.04 and Manjaro 20

TinLe, 2020-09-26:
See https://github.com/solokeys/solo-python/issues/92  the current udev rules for solo key is missing a few things.