solokeys/solo1 #466: Windows 10 FIDO2 NFC Compatibility

MichaelGrafnetter, 2020-08-16:
Hi, I have been playing with my new Solo Tap USB-A with firmware version 4.0 and I can't make it work on Windows 10 when it is connected through a [NFC reader from Feitian](https://www.ftsafe.com/Products/Card_Reader/Contactless). It always ends with an infinite loop of "New hardware found" and "Device disconnected" messages. FIDO2 keys from Feitian work seamlessly with this reader. Is there please an easy fix for this?

[image]

aseigler, 2020-08-17:
Is there a label on the back of the reader with any part numbers?  There were a series of NFC fixes that went in August 2019 (#280, #297) that sound very similar to the experience you describe, but if you are using 4.0 it has all of those patches already.

MichaelGrafnetter, 2020-08-17:
Hi @aseigler , thanks for your reply. No part numbers are listed, only these [specs on manufacturer's website](https://www.ftsafe.com/Products/Card_Reader/Contactless/Specification).

aseigler, 2020-08-17:
About the only "easy fix" you might try is having the Solo powered via USB while simultaneously using it on the NFC reader.  This will rule out a low power condition.  If that doesn't help (and I doubt it will), it's probably debugging time, comparing the request/response between the Feitian and Solo tokens with the WebAuthN operational event log or with USBPcap in Wireshark.

MichaelGrafnetter, 2020-08-17:
USB power did not help. 😒

kyleharte, 2021-01-26:
I'm assuming it's an issue with the way Windows handles FIDO keys. I'm assuming it could be the NFC timeout maybe

MichaelGrafnetter, 2021-01-30:
I just performed more NFC tests on Windows 10:
- Solo Key works well with ACS ACR1252 reader.
- Solo Key does not work with the Feitian reader (loop of connects and disconnects), while keys from both Crayonic and Feitian do work.

It would seem that Solo Key is only incompatible with that particular reader model.