solokeys/solo1 #469: Help Solokey hacker

parided, 2020-08-29:
Hi, I accidentally bought a hacker solokey instead of the standard solokey.
I have read that it can be converted to a standard solokey anyway.
I downloaded the 13 assets of version 4.0 from https://github.com/solokeys/solo/releases.
What is the flow chart to be able to make the solokey hacker a solokey standard?
I try to explain myself better. What is the order of the files to flash and what are they?

nickray, 2020-09-04:
You can't convert a hacker to a standard key, as the attestation private key is secret. It's possible to lock it down, is this what you want?

LEXOmx, 2020-09-06:
there is a text in solo repo: "Check out solokeys.com, for options on where to buy Solo. Solo Hacker can be converted to a secure version, but normal Solo cannot be converted to a Hacker version."

quoing, 2020-10-23:
I may have related question, I own hacker version of somu. Now I'm trying to access web that requires Level 1 Fido certification, but the web page claims that my version is not certified.. The same with locked version of somu hacker key.

Does it mean that hacker version is not certified? Or is issue on web page?

nickray, 2020-10-23:
The firmware is what's certified.

But what your web page is doing is checking the attestation certificate against a whitelist (or using the Fido metadata service to pull all known certified attestation certificates). An attestation certificate is valid for a specific attestation key, which needs to be kept secret. For this reason, hacker keys have a different ("well known") attestation key, and won't pass such requirements even if you lock the hacker key.

cole-h, 2021-01-23:
> [...] hacker keys have a different ("well known") attestation key, and won't pass such requirements even if you lock the hacker key.

Does this mean that some websites may reject a Hacker key?

Asking because, as we get closer to the Solo V2 launch, I'm wondering if I should pick up a Hacker to both hack on and use, or a normal version to use in addition to a Hacker to hack on.

quoing, 2021-01-24:
@cole-h My understanding is that Hacker version has different attestation key that is not part of **FIDO2 certification** (https://fidoalliance.org/certification/).. so if you need to authenticate to some services that requires key to be **FIDO2 Level1 certified** then you can't use hacker version.

There is no problem 2FA authenticating to common services - facebook, google, Linux PAM etc. all works with hacker version.

AngerM, 2022-02-26:
> @cole-h My understanding is that Hacker version has different attestation key that is not part of **FIDO2 certification** (https://fidoalliance.org/certification/).. so if you need to authenticate [...]

I don't think this is true anymore. I ordered 4 keys, 2 hacker and 2 normal for solo2. None of the websites (google/github/etc) accept the hacker keys.

hackerb9, 2022-03-08:
@nickray Is angerm correct: The hacker version of the Solo2 keys does not work for Google, Github, etc? That would be very disappointing as the hacker version of the Solo*1* keys had been working fine for me.

nickray, 2022-03-09:
None of these major websites check the metadata AFAIK. Plus Solo 2 is not certified, so there's no difference for them.

hackerb9, 2022-03-09:
Whew. I'm glad the hacker Solo 2 works. I look forward to purchasing my own soon.

AngerM, 2022-03-13:
@nickray have you tested? None of my hacker solo2 keys work with any websites I try. The standard solo2 keys work as expected. I've tried multiple sites and used the solo2 crate to update them to the latest firmware to be sure, but no such luck.

nickray, 2022-03-23:
That's a bit hard to confirm or deny. I can say that the hacker keys I have with the firmware I'll release work on Google + GitHub for sure. Once that update is out, if it still doesn't work please do open an issue for that.

AngerM, 2022-03-31:
Ok, I just re-checked and downloaded the newest 1.0.9 firmware and now ALL of my keys (hacker and non) work.

BalooRJ1, 2023-03-14:
I just got my key from backing the Indiegogo and I realized I ordered a hacker solokey. What do I need to update firmware wise to get it up to speed to use it with websites? Thank you.

hackerb9, 2023-03-14:
> I just got my Solokey V2 from backing the Indiegogo and I realized I ordered a hacker solokey. What do I need to update firmware wise to get it up to speed to use it with websites? Thank you.

There's nothing you need to do. My Hacker Solokey works fine with PayPal, Gmail, etc.