solokeys/solo1 #471: How to differenciate Solo from "Solo for hackers"?

juliomaranhao, 2020-09-15:
I ordered [1 solokey and 1 solokey for hackers](https://photos.app.goo.gl/fyKnZ7gQQVp4mMQS9). I received them and both works. But I didn't memorized which one is what type. I can see that one has v1.2 label while the other has v2.1 and an extra chip labeled N1. 

How do I differentiate then? Sorry to make this question. I should read the bags' label before opening them but I was excited. I just want to start using the solokey normally. "For hackers" solokey will be used as tests when/if I become hacker enough and have time. :-)

P.S.: I already called official support but I got this response from Haden Patrick: "You should be able to get device info through your system OS.  For a Mac, you can click on About this mac and then System Report, and look at the USB drive info for what is inserted.  Probably different for other computers!". And I replied with: "I use Win10Pro and I can see no way to differentiate both products by software. Can you check with the maker how to differentiate them?" As it is taking longer than I imagined I am posting here also.

P.S.2: I know a software called usbdeview. Unfortunately it shows no difference either.

ukscone, 2020-09-15:
Not 100% sure as i'm not at the right computer or even have a solo hacker to check but I think the solo-python repo has what you need. clone the repo, install and then one of the solo command options should tell you possibly version or verify

juliomaranhao, 2020-09-16:
> Not 100% sure as i'm not at the right computer or even have a solo hacker to check but I think the solo-python repo has what you need. clone the repo, install and then one of the solo command options should tell you possibly version or verify

Thanks. I installed python 3.8.6 RC1 x64 for Windows and issued a "pip install solo-python". I included my Python38\Scripts folder in PATH and started an admin Prompt.
Some tests with only v1.2 device connected:
```
C:\WINDOWS\system32>solo version
0.0.26
C:\WINDOWS\system32>solo ls
:: Solos
123456789012: Solo 4.0.0
```
"123456789012" is fake as I can't eval now if the original one is private info.
solo key [ wink | verify | version | rng hexbytes ] raises a solo.exceptions.NonUniqueDeviceError.
Example for wink subcommand:
```
C:\WINDOWS\system32>solo key wink
Traceback (most recent call last):
  File "c:\program files\python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\program files\python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\julio\AppData\Roaming\Python\Python38\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\solo\cli\key.py", line 460, in wink
    solo.client.find(serial, udp=udp).wink()
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\solo\client.py", line 47, in find
    p.find_device(dev=raw_device, solo_serial=solo_serial)
  File "C:\Users\julio\AppData\Roaming\Python\Python38\site-packages\solo\client.py", line 105, in find_device
    raise solo.exceptions.NonUniqueDeviceError
solo.exceptions.NonUniqueDeviceError
```

Now with only v2.1 connected:
```
C:\WINDOWS\system32>solo ls
:: Solos
123456789021: Solo 3.1.2
```
Same solo.exceptions.NonUniqueDeviceError for "solo key ...".

Is "solo ls" output enough to discriminate them? I hope so. Cheers.

nickray, 2020-09-18:
The way to do this currently is to run `solo key verify`. This generates a credential, that is signed by the attestation key of the device.

- if it's an authentic secure Solo, this will be the official SoloKeys attestation key, of which the public key is well-known, and the private key a trade secret (well, until someone hacks our keys...). So if the signature by the attestation key over the generated ephemeral credential is checked to be valid (which is what `solo key verify` does), and you trust our supply chain, you can trust that you have an authentic secure Solo.

- if it's not, then there are a couple options. Hacker keys with our unmodified firmware use our hacker attestation key, of which both private and public key are well-known. But since the private key is public, this could also be some random other key, even of a different vendor. On the other hand, if it's a hacker key, you or someone else might have changed the attestation key altogether.

So to sum this up, you can determine "secure" vs "something else" via `solo key verify`.

Once done, I suggest you pick and remember case colours to distinguish them :)

juliomaranhao, 2020-09-21:
Thanks for all help. I decided to repeat the tests in Linux and bingo!

I used the following tools:

1. Debian 10 PC;
2. Udev rules from Yubico;
3. python3 -m venv solotest
4. (inside venv) pip install -U pip
5. (inside venv) pip install solo-python

```
(solotest) $ # v2.1 device
(solotest) $ solo key version
3.1.2 locked
(solotest) $ solo key verify --pin XXXX
Valid Solo Tap with firmware from SoloKeys.

(solotest) $ # Now with v1.2 device
(solotest) $ solo key version
4.0.0 unlocked
(solotest) $ solo key verify --pin XXXX
Solo Hacker firmware.
```

No exceptions raised! This issue can be closed!

> Once done, I suggest you pick and remember case colours to distinguish them :)

Just did it. Cheers.

closed 2020-09-21 by juliomaranhao

AWHubGit, 2021-05-02:
Helpful