solokeys/solo1 #472: Standalone installation - RuntimeError: No FIDO device found

Peter98, 2020-09-16:
Hi,
I am trying to run the stand alone build using instructions under the sub heading "Developing Solo (No Hardware Needed)" (https://docs.solokeys.dev/repo-readme/#developing-solo-no-hardware-needed).
When running ./main i get the following:

[DEBUG] Using UDP backing
state file exists
[ERR] Current firmware version address: 0x556276e2e110
[ERR] Current firmware version: 4.0.0.0 (04.00.00.00)
[STOR] Auth state is initialized
[STOR] pin not set.

And when running the tools code I predominately get the error:

RuntimeError: No FIDO device found

I am not sure if I am missing build instructions, I am not using any virtual environments just running the code as described in the above link.

Any help is appreciated,
Peter

nickray, 2020-09-18:
I assume the tools you use are trying to use the USB interface, and not the UDP interface used by the PC build?

Lethargo, 2020-09-23:
I am using these tools: https://github.com/solokeys/fido2-tests 

And running this command: pytest tests/standard -s

Yarflam, 2021-07-12:
I want to try ... but it's not working too.