solokeys/solo1 #476: Python NonUniqueDeviceError with multiple Solo Taps plugged in

Trapacid, 2020-10-08:
solo-python throws an error when there are more than one security keys are plugged in:

```python
Traceback (most recent call last):
  File "/Users/ashley/homebrew/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/solo/cli/key.py", line 472, in ping
    client = solo.client.find(serial, udp=udp)
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/solo/client.py", line 47, in find
    p.find_device(dev=raw_device, solo_serial=solo_serial)
  File "/Users/ashley/homebrew/lib/python3.8/site-packages/solo/client.py", line 105, in find_device
    raise solo.exceptions.NonUniqueDeviceError
solo.exceptions.NonUniqueDeviceError
```

Issue occurs on macOS Big Sur using the USB-A and USB-C variants of Solo Tap, both plugged in at the same time. The issue occurs running `solo key ping` or `solo key wink`. May happen with other commands.

The same error also occurs on Fedora 33 with a Yubikey 5 NFC and a Solo Tap USB-A NFC.

schlomie, 2020-10-08:
If multiple keys are attached, you will need to specify the intended key by its serial number.

To get the serial #:
```bash
$ solo ls
:: Solos
123456789012: SoloKeys Solo 4.0.0
```

Then issue the command:
```bash
$ solo key wink -s 123456789012
```

Try
```bash
$ solo key wink --help
Usage: solo key wink [OPTIONS]

  Send wink command to key (blinks LED a few times).

Options:
  -s, --serial TEXT  Serial number of Solo to use
  --udp              Communicate over UDP with software key
  --help             Show this message and exit.
```

Trapacid, 2020-10-08:
Would help new users if that was told to the user when that issue happens.

closed 2020-10-08 by Trapacid