solokeys/solo1 #479: Could the solo key be disabled if the wrong firmware was uploaded?

LockRing, 2020-10-20:
Hello,
I've read a bit of code and it seems to validate after flash the firmware when I update the key. Is that mean solo key cannot boot if someone tries to update with older or malformed firmware?(solo client will check before update. anyway)
In this case, can the key be recovered with the latest official firmware update? Nothing has changed with the update except for the application segment, so I think it's possible.

nickray, 2020-10-20:
What happens is this:
- we publish firmware + signature for hash of firmware
- updater flashes new firmware
- meanwhile, hash of new firmware is calculated
- at the end, calculated hash is compared with hash for which we published signature

same hash => firmware is activated, device reboots into new firmware
different hash => device remains in update mode

Different hash can happen either if flashing was unsuccessful (e.g. aborted), or the firmware was tampered with.
In this case, you need to update again, either with the same firmware if the update aborted, or a proper firmware, if it was incorrect.

So yes, you're just stuck in limbo for a while.

LockRing, 2020-10-20:
Thanks for the answer. I just wanted to know what happens if wrong firmware uploaded because existing firmware is gone when update the firmware anyway.
It would be better if the existing firmware remained until the uploaded firmware was verified(something like android A/B), but it seems difficult with the current solo key storage space.

closed 2020-10-20 by LockRing