solokeys/solo1 #480: Security concerns about seed the solo key

LockRing, 2020-10-22:
I saw the [dicekeys](https://dicekeys.com) and read the [specification](https://github.com/dicekeys/seeding-webauthn). This allows you to replicate the solo key if you previously seeded it. 

Not implemented yet, but I think this could be vulnerable in some situations. For example, if someone secretly seeded and replicated your key, that person could go through the 2FA you registered from that time. It's hard to notice until breach happens.

Many FIDO 2FA enabled sites allows you to register multiple keys for backup, it would be more secure than allowing key replication.

Is there any way planned to prevent setting seed?

conorpp, 2020-10-23:
Yes, the multiple key backup would be the more secure way since you can be assured against no duplicate keys, and that will remain the default.  SoloKeys distributed secure keys will not have the backup capability for this reason.  Backup will only be an option to people using hacker keys, or keys distributed from the DiceKeys campaign.  So setting a seed starts with deliberately getting a device designed to be provisioned, rather than a "normal" one.

LockRing, 2020-10-23:
It looks fine. Then cross-firmware update should be rejected. Maybe create another pair of signing keys and only use them for SoloKeys distributed from the DiceKeys campaign?

back2root, 2020-10-23:
For this kind of attack to be carried out successful it is required, that Mallory seeds a solo key before Alice registers this key as an authenticator at a relying party.

To prevent an attack like this, Alice can just reset or seed her key after receiving and before registering the key the first time.

When registering the key for additional relying parties, resetting the key is not a valid option as it would destroy existing registrations with other relying parties.
Instead Alice can just authenticate to an relaying party, where she has registered her key previously. If authentication succeeds, Alice can assume that Malory has not seeded her key in the time between registration and authentication. Right after that Alice should be safe registering her key at additional relaying parties.

LockRing, 2020-10-24:
@back2root
> For this kind of attack to be carried out successful it is required, that Mallory seeds a solo key before Alice registers this key as an authenticator at a relying party.

You're right. That's why I said in some situations. What I was most worried about was seeding the secure SoloKeys currently on sale. Most people who only buy SoloKeys will think that the keys shouldn't be duplicated. It is possible to duplicate at the hardware level, but if that happens, noticeable changes will occur(missing for several days and chip scratches).
Reset or seed immediately after receiving the key is a good idea, but many don't. And we can confirm that the key is not seeded the way you said, however, there is no guarantee that everyone will follow these instructions. Therefore I'm satisfied to be able to seed only the SoloKeys distributed with the DiceKeys. Anyone buying DiceKeys implicitly agrees that keys can be duplicated and is probably the one who wants it.

Fuse or something like that would be a good option for DiceKeys users. The only restriction with that is having to buy a new SoloKeys if the dice need to re-roll it. Anyone who sets that option will know what they are doing, so it won't be a problem.

My1, 2020-11-12:
also obviously if someone gets your secret, you are screwed and have no way to even know.

what might be useful would be some kind of key derivation that would allow to automatically correlate the keys together but the individual keys still being different enough to allow services to notice the new key and notify the user. one could for example run the follwoing structure, which the relying parties would need to adapt, but I thought about it in a way that doesnt interfere with FIDO2 or U2F itself in the slightest.

There is a standard called BIP32 in the cryptocoin scene that has some very interesting levels of key derivation including public to public key derivation. so one could use an extended public key (also known as xpub) in the following way:

1) the signer of the attestation certificate is derived from the xpub, and contains some basic info like the derivation path and some kind of hash of the xpub (we want to avoid just giving out the xpub to everyone).
2) the keys for the attestation certificate on the device itself is generated at random and signed by the above. (this is to make sure that there are no private keys from the derivation on the device, due to the ability that with the parent pubic and private child keys to generate the parent private key (which we dont want)
3) the Master Keys of the device itself are again random, and attestation builds the link to the above.

now, at a relying party that would support this style, you could register the xpub, and when you lose the device the following would happen:

1) you would go to the RP and go to the login page with some link to apply a replaced key (similar to a forgot password page)
2) you would enter your username or whatever is your "login identity" so that the service knows who to look for
3) you go through a normal U2F/FIDO2 register process making sure to keep attestation data intact
4) the RP can build the chain
`user Authorization -> xpub -> attestation signer -> attestation cert -> device keys`
and thereby link the new device back to the user, and put it into the list, at that point logging in the user and notifying him/her about the event
5) the key used originally can then be revoked by the user if lost or stolen to prevent it from being used again, or left alone if just for example misplaced at home

back2root, 2020-11-12:
From my point of view, the idea suggested by @My1 is more an enhancement request for the FIDO2 standard, not a thing a single Authenticator should implement on it's own.
Reason behind this is because the RPs need to actively support this suggestion.