solokeys/solo1 #484: Authenticator Level 3?

hasufell, 2020-11-06:
Afais solokeys are [level 1](https://fidoalliance.org/certification/authenticator-certification-levels/authenticator-level-1/) only. Are there any plans to support [level 3](https://fidoalliance.org/certification/authenticator-certification-levels/authenticator-level-3/)?

Trapacid, 2020-11-06:
If I understand correctly, this can't be accomplished, specifically in the hardware area.

Yubikey firmware is baked into it as it's being manufactured and **cannot** be upgraded without buying a new one. Solokeys offers the ability to update the firmware even after manufacturing.

You also cannot access the PCB of the Yubikey without using a chemical substance to eat away the material, as described here in this teardown of the Yubikey 5 NEO: http://www.hexview.com/~scl/neo5/
The PCB of the Solokey can be accessed freely.

conorpp, 2020-11-07:
This isn't a goal for Solo "v1" but we are considering it for the next hardware model we're working on.

closed 2020-11-07 by conorpp