solokeys/solo1 #487: I forgot my pin and now my key is blocked

emdouglas2001, 2020-11-18:
Hi,
 As the subject says, I forgot my pin and now my key is blocked, is there way to unblock it.

Many Thanks

jolo1581, 2020-11-19:
Use solo python.

I think the command is:
solo key reset

But this will also delete all credentials and connected sites

conorpp, 2020-11-20:
Yes for any FIDO key this means you need to reset it, and will loose all of your credentials.  You can use our solo-python client to do so, or use the Security Key tool in Chrome settings.  You can also do it in Windows security key settings.

closed 2020-11-20 by conorpp

emdouglas2001, 2020-11-20:
Thanks Guys,
That sorted my problem.
Many Thanks from a much happier Scotsman

svn2208, 2022-03-07:
Is it possible to remove the pin again?

nickray, 2022-03-07:
You can only remove the PIN by resetting it.