solokeys/solo1 #488: I think my key is dead

victorlane, 2020-11-19:
I put in my key into my desktop and windows says it's setting up the key but the light doesn't blink. I put it in my laptop and nothing happens at all, no activity.

victorlane, 2020-11-19:
Just ran solo ls using solo-python and it doesn't show up

conorpp, 2020-11-19:
Since it sounds like it's enumerating, can you provide any information on the USB descriptor that shows up?  Like `lsusb` on Linux, or a device in the device manager on Windows.

victorlane, 2020-11-19:
> 
> 
> Since it sounds like it's enumerating, can you provide any information on the USB descriptor that shows up? Like `lsusb` on Linux, or a device in the device manager on Windows.

[image]

I did however find something I didn't know what it is called "STM32 Bootloader", not sure if it's anything to do with Solo.

victorlane, 2020-11-19:
Only thing I could find that might have something to do with Solo is a "STMicroelectronics STM Device in DFU Mode" BUS Device

conorpp, 2020-11-19:
The key has not been programmed.  Did you try reprogramming it or receive it as is?  If you got it like this, please contact the distributor or hello@solokeys.com.

closed 2020-11-19 by conorpp

victorlane, 2020-11-19:
I got it like this, can I reprogram it myself?

conorpp, 2020-11-19:
Yes, starting from step 2:
https://docs.solokeys.io/programming/#updating-a-hacker-to-a-secure-solo

Use [bundle-secure-non-solokeys-4.0.0.hex](https://github.com/solokeys/solo/releases/download/4.0.0/bundle-secure-non-solokeys-4.0.0.hex) or [bundle-hacker-4.0.0.hex](https://github.com/solokeys/solo/releases/download/4.0.0/bundle-hacker-4.0.0.hex).

victorlane, 2020-11-19:
> Yes, starting from step 2:
> https://docs.solokeys.io/programming/#updating-a-hacker-to-a-secure-solo
>
> Use [...]

Trying to reprogram it using
solo program dfu bundle-secure-non-solokeys-4.0.0.hex & it raises an error `No Backend Available`

conorpp, 2020-11-19:
Need to install libusb.  A bit tricky on windows, easiest to do on Linux or MacOS.

victorlane, 2020-11-19:
Got it to work on Windows, bit odd that they sent out a non-programmed key though. Anywho, thanks for the support & patience